raspberrypi/pico-feedback #99: RP2040 Datasheet, the PICOBOOT USB control requests' request types are incorrect

yvt, 2021-02-18:
In "2.8.5.5.1. INTERFACE_RESET (0x41)", the datasheet says it's `00100001b` (interface + class + out), but [it's actually `01000001b` (interface + vendor + out)](https://github.com/raspberrypi/pico-bootrom/blob/00a4a19114195e20fb817bdfbca1165e157eef37/bootrom/usb_boot_device.c#L217-L237). Similarly, "2.8.5.5.2. GET_COMMAND_STATUS (0x42)" also shows an incorrect value of `bmRequestType`.

iosabi, 2021-08-29:
I found the same issue. In addition, the GET_COMMAND_STATUS (0x42) is an [input command](https://github.com/raspberrypi/pico-bootrom/blob/00a4a19114195e20fb817bdfbca1165e157eef37/bootrom/usb_boot_device.c#L220) so the `bmRequestType` should have the MSB set: `11000001b`, and it returns 16 bytes so `wLength` should be 16 bytes.

To summarize, `bmRequestType` should be:
 * `01000001b` for 2.8.4.5.1. INTERFACE_RESET (0x41)
 * `11000001b` for 2.8.4.5.2. GET_COMMAND_STATUS (0x42), in addition `wLength` should be changed to `0010h` here.

kilograham, 2021-09-13:
fixed internally

closed 2021-09-13 by kilograham