raspberrypi/pico-feedback #103: Internal Temperature Sensor readings are too high

Noxdor, 2021-02-22:
Testing a program on my new pico, trying to read the temperature from the internal temperature sensor, the results coming back are clearly too high for my environment.

The code I'm running:
````
#include <pico/stdlib.h>
#include <hardware/adc.h>

stdio_init_all();
adc_init();
adc_set_temp_sensor_enabled(true);
adc_select_input(4);

// 12 bit conversion factor
const double conversion_factor = 3.3f / (1 << 12);

while (true) {
                double result_tmp = adc_read() * conversion_factor;
		double tmp = 27 - (result_tmp - 0.706) / 0.001721;  // Formular from the Pico  C/C++ SDK  Manual
		printf("Temperature: %4.2f\n", tmp);

		sleep_ms(995);
}
````

I'm getting readings and they're changing, the values are just clearly too high. Room temperature is around 20 - 25 deg C.

Is it an error in the code or a problem with the SDK?

Noxdor, 2021-02-26:
Chip temperature can't be a factor after just starting up the chip and furthermore not running intense computation. Over a longer period of time, of course, that might come into play.
"Clearly to high" is in my understanding a deviation of around 10-15 deg. I just saw I didn't mention the shown data from the pico, sorry, it was showing 30 - 35 deg. That is a difference of up to 10 deg. As soon as I find time, I'll try to measure the room temperature more precisely with a different thermometer and adjust the mapping equation of the pico. Then I'll port the code to my other two Picos and see, what they'll say.

Noxdor, 2021-02-26:
Watching this video (https://www.youtube.com/watch?v=dsu-c6kSIe0&ab_channel=LearnEmbeddedSystems) I understand it's really a general problem of the pico and can go in both directions. 

While my pico is 10-15 deg too high, his is 10 too low. It really must come from manufacturing differences. 

The sensor is anyways of course not perfectly precise, but the systematic measurement uncertainty is clearly varying drastically between picos. So either

1. Invest some time to properly calibrate the sensor by measuring the temperature with a finer sensor at different temperatures while also measuring the pico sensors voltage. Then you can adjust the slope and offsets in the given temperature function 
`27 - (voltage - 0.706) / 0.001721`.
2. Get an external sensor, as you would do with any other microcontroller.

I hope I could help someone struggling with the same issues.

closed 2021-02-26 by Noxdor

polarbearing, 2021-07-05:
The calculated chip temperature depends critically of the actual Vref value being used in the calculation.

The code example uses ```conversion_factor = 3.3f```, i.e. 3.3 V from the Vdd, however the data sheet teaches us that the 200R resistor R7 causes a voltage drop of 30 mV from Vdd, thus the nominal Vref is only 3.27 V. The actual value may vary, so I measured Vdd(real)=3.258 V and Vref(real)=3.22 V (drop 38 mV). 

Applied to the same raw ADC reading of **0x379 = 889** decimal, measured at room temperature of 27 C, we get a wide range of temperatures calculated:

Vref conversion factor  | 3.3 | 3.27 | 3.258 | 3.22
-- | -- | -- | -- | --
ADC |0x379|0x379|0x379|0x379
Voltage(V)|0.7162 | 0.7097 | 0.7071 | 0.6989
Temp(C)|21.053 | 24.836 | 26.349 | 31.142

For a chip temp, the highest of them seems plausible. 

Thus the only real solution is to get an LM4040 tomorrow and clamp the Vref down to 3.00 V.

aallan, 2021-07-05:
Yes, the onboard temperature sensor is very sensitive to the reference voltage. See the latest edition of the [RP2040 Datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf). 

[image]

polarbearing, 2021-07-05:
Indeed that is where I got my wisdom from. The issue with the ~30 mV voltage drop is from the [Pico datasheet](https://datasheets.raspberrypi.org/pico/pico-datasheet.pdf): 
[image]

Reading those datasheets turned out to be more productive than reading dozens of application examples each copying their faults from the other.

JamesH65, 2021-07-10:
It's almost as if the engineers that designed the chip wrote the datasheet!

Michael-F-Ellis, 2021-08-06:
Thanks to all for the clarifications.  I was seeing very large errors (reporting 81°C when the air temp was 26.9°C when measured beside the pico with a precision sensor). 

A crude 1 point calibration against the precision sensor got the temperature reports within 0.5°C without using an external ADC reference input.

The good news is that the raw reports from the ADC are stable +/- 1 LSB.