raspberrypi/pico-feedback #104: How to switch the CPU clock to its maximal value

omikron88, 2021-02-24:
Hello,
How to switch the CPU clock to its maximal value 133 MHz ?

daveythacher, 2021-03-02:
Is the max 125MHz, 133MHz or 150MHz?

omikron88, 2021-03-02:
The datasheet says 133MHz

aallan, 2021-03-02:
> Is the max 125MHz, 133MHz or 150MHz?

As noted in the datasheet the maximum clock frequency is 133MHz, the SDK default frequency is 125MHz.

daveythacher, 2021-03-02:
> > Is the max 125MHz, 133MHz or 150MHz?
> 
> As noted in the datasheet the maximum clock frequency is 133MHz, the SDK default frequency is 125MHz.

The PLL and Cortex speed is 133MHz. (1.2 and 2.18.1) The clk_sys lists a max nominal of 125MHz. (2.15.3.1) Power consumption shows overclocking to 150MHz. (5.4.1)

I know the answer it will likely work at 133MHz, but may be designed to work at 125MHz. I also know that overclocking may allow higher clock speeds.

WestfW, 2021-03-03:
There is `set_sys_clock_khz(rate, required)`
This is apparently considered part of pico_stdlib, but perhaps that should be mentioned in the hardware_clocks section as well?

It would be nice if the SDK's default system clock were mentioned more clearly somewhere.  Currently, the only place I could find this was buried in the Logic Analyzer example.

closed 2021-05-25 by aallan