raspberrypi/pico-feedback #106: Typo in Python SDK build-date: 2021-02-01 plus suggestion for improvement of text build-date: 2021-02-23

PeterPotgieser, 2021-02-25:
I think there is a typo in https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf - build-date: 2021-02-01. Section 4.1.1 says '_Then open up the Run menu and select Run →Select Interpreter, picking "MicroPython (Raspberry Pi Pico)_" from the drop down, see Figure 4.'
But I think it should read _'open up Tools -> Options -> interpreter tab_' (to get to the screen you show in figure 4), at least that how it goes in my Thonny 3.3.3 on Pi _4. 

I noticed there is a new version (build-date: 2021-02-23) of the document. The latter now, in 4.1.1, refers to a button at the bottom right to achieve the selection ... If you're running Thonny 3.3.3 on Pi 4 using VNC on Windows 10 like me,  the bottom right button is not normally clearly visible. Hence the description on how to select the interpreter in the previous build of the   document worked better, apart from the typo as mentioned.

lurch, 2021-02-25:
I'm gonna assign this to @aallan since he was the last person to edit those instructions :wink:

aallan, 2021-02-25:
The Thonny developer has told us that the bottom right button is the preferred method, and the menus are subject to change.

closed 2021-02-25 by aallan