raspberrypi/pico-feedback #107: RP2040 GPIO maximum speed

EngineerWill, 2021-02-27:
I have checked the data sheet of RP2040, but I can't find the highest frequency that GPIO can get
In Figure 37 i see slewrate, but no action description was found. Only full speed and low speed are available.

daveythacher, 2021-03-02:
I mentioned this earlier and a few other things not mentioned here, but it got removed.

In terms of frequency you will likely have a decent amount of performance. The quality of the signal may be questionable. The idea here is you use current and/or skew control to adjust it.

The Pi never really documented this as I recall. It did mention rise and fall times. There is a few specs here that could change things.

Best guess is 50 to 66.5MHz. 

I am not sure what the max speed is 125 or 133. The PLL and CPU Core are capable of 133 but system clock is listed at 125. I could not find ssi_clk. However most peripherals appear to support half of system clock.

GPIO Mux is not really well defined. SIO suggests single cycle performance, which would be CPU core clock. However it could run into a slower clock tree which limits the speed. GPIO Mux output performance is not listed. However input performance is listed at 50MHz.

The pin mapping allows SPI on nearly all pins. However GPIO IN/OUT is reserved. SSI is reserved also. So likely all GPIO pins have the same rough performance. 

It could support 133MHz, but I doubt that will have decent wave forms. The CPU core and PIO can run at full speed.

In summary I am not sure it matters. This may get deleted also.

Likely this is compatible with the Pi. So if you used the Pi GPIO drivers before this will likely be the same.

EngineerWill, 2021-03-02:
### About output

In the process of testing the VGA interface, I set the system clock to 148.5MHz, the GPIO output frequency seems to reach half of the system clock（74.25Mhz）, but the specific waveform cannot be detected by my existing oscilloscope.

EngineerWill, 2021-03-02:
But I don't know what Slewrate does in Figure 37, and what is the function and significance of low Rate

daveythacher, 2021-03-02:
> But I don't know what Slewrate does in Figure 37, and what is the function and significance of low Rate

http://www.interfacebus.com/IC_Output_Slew_Rate.html
https://en.wikipedia.org/wiki/Slew_rate

lurch, 2021-03-02:
> I could not find ssi_clk.

Section 4.10.2.1. in the dastasheet says "Clock connections are as follows: pclk and sclk are driven from clk_sys"

lurch, 2021-03-02:
I'm not a hardware-person, but the RP2040 _can_ do very fast GPIO, see https://github.com/Wren6991/picodvi for an example.

EngineerWill, 2021-03-06:
> I'm not a hardware-person, but the RP2040 _can_ do very fast GPIO, see https://github.com/Wren6991/picodvi for an example.

Well, I guess my problem is probably solved, but I still want a definite number.

By the way, can you tell me why the Mircopython repository is missing

Looking forward to your reply

aallan, 2021-03-06:
> By the way, can you tell me why the Mircopython repository is missing

Missing? The RP2040 MicroPython port has now been merged upstream to the main MicroPython repo, see micropython/micropython#6791. Going forward you should report problems with the port  there, https://github.com/micropython/micropython/, and the local fork of the repo has been deleted as it is no longer needed. [Instructions](https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf) on how to build MicroPython from source have been updated accordingly.

Technical support for MicroPython can be found on the MicroPython forums, https://forum.micropython.org/viewforum.php?f=21.

EngineerWill, 2021-03-06:
> Missing? The RP2040 MicroPython port has now been merged upstream to the main MicroPython repo, see [micropython/micropython#6791](https://github.com/micropython/micropython/pull/6791). Going [...]

Thank you very much for your reply

closed 2021-03-06 by EngineerWill

lurch, 2021-03-06:
> > I'm not a hardware-person, but the RP2040 _can_ do very fast GPIO, see https://github.com/Wren6991/picodvi for an example.

> Well, I guess my problem is probably solved, but I still want a definite number.

In internal discussions, one of the hardware-people said that this is very hard to characterise, as it's dependent on so many external factors (capacitance/impedance/etc.), that any "definite number" we gave would probably be meaningless. (and people would probably complain when their "real-world circuits" don't match up with our "ideal-setup measurements"? :shrug: )

And unless you're using PIO, the "maximum GPIO speed" achievable with MicroPython will likely be a lot lower than with C.

daveythacher, 2021-03-06:
> > > I'm not a hardware-person, but the RP2040 _can_ do very fast GPIO, see https://github.com/Wren6991/picodvi for an example.
> 
> > Well, I guess my problem is probably solved, but I still want a definite number.
> 
> In internal discussions, one of the hardware-people said that this is very hard to characterise, as it's dependent on so many external factors (capacitance/impedance/etc.), that any "definite number" we gave would probably be meaningless. (and people would probably complain when their "real-world circuits" don't match up with our "ideal-setup measurements"? 🤷 )
> 
> And unless you're using PIO, the "maximum GPIO speed" achievable with MicroPython will likely be a lot lower than with C.

This is true. Normally what they do is list a test condition. Single ended rail to rail square wave will likely only be possible at lower frequencies. There is a lot that goes into it. Pin cap, trace cap, trace impedance, rise/fall times, etc.

It would depend what you are trying to do with it. However you may find success with external drivers. However if you over drive you can cause ringing. However there is a number of tricks like slew rate, drive strength, termination, etc. which can help.

The documentation could document the max clock possible by the hardware without concern for the pin driver. In this case square waves may become sine waves with less amplitude. This would be bad for certain signaling schemes but may not be critical for others. DVI/HDMI uses current mode logic with differential signaling.

daveythacher, 2021-03-06:
> This is true. Normally what they do is list a test condition. Single ended rail to rail square wave will likely only be possible at lower frequencies. There is a lot that goes into it. Pin cap, [...]

They could list every parameter within a range of tolerance. They could list example conditions and the corresponding performance. The IO design and high frequencies change things a little. It reminds me of a FPGA where there is a ton of bandwidth if you do not do something silly.

In summary, listing a frequency is possible. However confusion is possible. Indirectly we know what the max frequency possible. However there is not a lot of information for generating expectations about the signal. This is very technical and potentially confusing.

The argument could be made that you design for the high performance signal. You either get it or you don't. However the design assumptions are a possible source of error. However I do not think this changes anything significantly. This is a really terrible answer but for beginners and more casual users this will likely not be a huge concern. 

I think an argument for releasing the total list of parameters could be made for the more advanced users. However I think I would make that it is its own release. Assuming the Pi foundation is willing to release that or has those numbers for the entire production line.