raspberrypi/pico-feedback #108: Windows instructions for building OpenOCD for Pico SWD incomplete

e107steved, 2021-02-27:
I've just tried to build OpenOCD for Windows (W7 64-bit in my case) and found a number of issues with Appendix A of "Getting started with Raspberry Pi Pico" build date 23-02-2021. Possibly a reordering, plus a few extra comments.

Note: I went straight to Appendix A without doing anything else that might have been mentioned earlier in the docs.
1.  When installing MSys I had a number of repositories which failed to download the first time (probably SourceForge was down). Recoverable, but might be worth a note, especially since MSys seemed to think it had succeeded, but the absence of the reqired functionality generated errors later on, and I had to use educated guesswork to work out what to reinstall.
2. The rest of the compile process worked OK (although can't say I like working directly on a cloned repository, but that looks like OpenOCD's choice, not yours) until the "test your build" - which does indeed error out, but because it can't find  libusb-1.0.dll - that's not installed until later!
3. I then went to download a .uf2 file to flash Picoprobe, but couldn't find it anywhere. There is no "utilities" section on the linked "getting started" page, and a search for "debugging" and "picoprobe" also fails. Not found that file yet.
4. I then moved on to install the USB driver using Zadig; but no Picoprobe option displayed, presumably because one must be plugged in to be found.

e107steved, 2021-02-27:
As follow-up, I found the .uf2 in the build directory of Picoprobe in the SDK.
Zadig does indeed need the Picoprobe to be plugged in; perhaps a note that it (I assume, since its installing a driver) needs to be run as administrator.

aallan, 2021-03-01:
> I then went to download a .uf2 file to flash Picoprobe, but couldn't find it anywhere. There is no "utilities" section on the linked "getting started" page, and a search for "debugging" and "picoprobe" also fails. Not found that file yet.

Go to the second tab of https://www.raspberrypi.org/documentation/rp2040/getting-started/ and scroll down.

[image]

closed 2021-05-25 by aallan