raspberrypi/pico-feedback #109: Documentation: .sideset (opt) bit stealing information is wrong

alwynallan, 2021-02-28:
Both the **RP2040 Datasheet** (p. 338) and the **Raspberry Pi Pico C/C++ SDK** (p. 49) documents say 

> Additionally opt may be specified to indicate that a side <value> is optional for instructions (not using this requires stealing an extra bit - in addition to the <count> bits - from those available for the instruction delay).

which is wrong. Removing "not" would make it a true statement.

lurch, 2021-02-28:
I wonder if it was _supposed_ to say "(note using this..." ? :shrug:

lurch, 2021-03-30:
#131 agrees that the `not` here should be `note`, and no-one has said otherwise. So I just went to make this change, and see that @Wren6991 has already beaten me to it! :rofl: Thanks Luke :+1:

closed 2021-03-30 by lurch

aallan, 2021-03-30:
Will be in the next PDF docs release. 😃