raspberrypi/pico-feedback #110: Documentation: USB sie_status CONNECTED (bit 16) should be R-WC instead of RO

hathach, 2021-03-02:
RP2040 Datasheet v1.2 Table 415. USB: SIE_STATUS Register page 413, CONNECTED (bit 16) is both WC and read-only making it R-WC. Read to determine the current state of connection (connected/disconnected), clear to ack the interrupt event.

[image]

The hint is at USB INTS register 

[image]

liamfraser, 2021-03-02:
Thanks. @lurch is going to look at this.

aallan, 2021-05-25:
@lurch Did you manage to take a look, or is it still on backlog?

lurch, 2021-05-25:
Still on backlog.