raspberrypi/pico-feedback #111: nmake fail to compile from 2nd times

brtchip-tuannguyen, 2021-03-03:
On window, after compiled blink example successful, nmake return error when run again:

[image]

Look at the compiler_depend.make and see that some locations are modified to incorrect string: Some location start with "D:\\" were changed to "D\\:"

[image]

There is a work around to compile, that is correct the location in compiler_depend.make (with notepad/vim...etc), then run nmake.

This issue happens on VScode too.

lurch, 2021-03-03:
Duplicate of https://github.com/raspberrypi/pico-sdk/issues/169 ?  What version of CMake do you have installed?

brtchip-tuannguyen, 2021-03-04:
> What version of CMake do you have installed?

I using cmake version 3.20.0-rc2

lurch, 2021-03-04:
As I explained in that issue I linked to (and as also explained in section 9.2.1.2. of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf) there seems to be a bug in that version of CMake. Please uninstall it and try using CMake 3.19.6 instead.

closed 2021-03-04 by aallan

aallan, 2021-03-29:
Reported upstream as a CMake bug, https://gitlab.kitware.com/cmake/cmake/-/issues/21997, by @lurch.

lurch, 2021-04-12:
FYI this is now fixed in [CMake 3.20.1](https://cmake.org/download/) :tada:

aallan, 2021-04-12:
> FYI this is now fixed in [CMake 3.20.1](https://cmake.org/download/) 🎉

Can you update the NOTE block in the PDF documentation to indicate that, we should leave the warning in as there will be versions of CMake floating around which are broken in this regard for a long time, but the exact wording now needs updating I think.