raspberrypi/pico-feedback #112: Suggestion to avoid 'scattered' information

PeterPotgieser, 2021-03-03:
Hi, I am afraid it is my fault ,,, but after more than an hour looking through various pico related publications (.pdf's) I seem not to be able to find an answer to the simple questions: "what is the current that any GPIO pin can source in condition '1' (3.3V?) and what is the current that any GPIO pin can sink in condition '0' ? How many pins can source / sink simultaneously ?"

I am from the TTL era, where this was (almost) always specified for every digital input / output, terms like 'fan-in' and 'fan-out'. (see e.g. https://www.quora.com/What-is-fan-in-and-fan-out-in-logic-circuits)

I suggest this basic information be easily retrievable like on the page 4 of https://datasheets.raspberrypi.org/pico/pico-datasheet.pdf where it says '30 multi-function General Purpose IO (4 can be used for ADC) -   1.8-3.3V IO Voltage (NOTE Pico IO voltage is fixed at 3.3V)'. Information that could be added here could be of the format:
logic '1' = x.y V DC, max. output current z mA
logic '0' = a.b V DC, max. input current c mA
I suspect that the total current in to / out of the RP2040 via the GPIO pins is also subject to a maximum, but I am afraid I could not find that info either. Apologies if I turn out to have overlooked it ....
--

PeterPotgieser, 2021-03-03:
? Don't know why the font in my message changed ?

aallan, 2021-03-03:
> ? Don't know why the font in my message changed ?

Fixed it for you. :)

aallan, 2021-03-03:
See #37 for discussion, and Section 5.2.3.4. and Table 630 of the current [RP2040 Datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf).

closed 2021-03-03 by aallan

JamesH65, 2021-03-03:
@aallan @lurch Isn't this data in the next version of the datasheet, and quite soon to be published? Or was that something else?

aallan, 2021-03-03:
> @aallan @lurch Isn't this data in the next version of the datasheet, and quite soon to be published? Or was that something else?

You posted while I looked it up. :)

PeterPotgieser, 2021-03-04:
Thanks for the answer. I followed your suggestions and found the information. From a technical point you are correct. From a user point of view, I think my suggestion still stands:

If a user buys a pico to start embedding it and programming it using e.g. the Python SDK, I think the information in https://datasheets.raspberrypi.org/pico/pico-datasheet.pdf should be able to provide what the user needs to know without having to scan almost 700 pages in a datasheet of the chip itself. Besides, how is this user supposed to know that the chip’s GPIO characteristics are those of the pico board itself ? There could have been some buffering done ? I suggest that, for these users, the essentials of Table 630 are included in the PICO datasheet.

lurch, 2021-03-04:
The Raspberry Pi Pico board can be viewed as a "breakout board" for the RP2040 chip, so the Pico datasheet needs to be read in conjunction with the RP2040 datasheet. (and IMHO the maximum current that can be sourced / sunk by a GPIO pin is definitely a chip-level spec rather than a board-level spec :wink: And you only have to scroll down to page 7 of the RP2040 datasheet to see the "Electrical and Mechanical" section mentioned in the Table-of-Contents.)
The trouble with duplicating information across multiple datasheets is that it's very easy for those different parts to get out of sync with each other.

PeterPotgieser, 2021-03-04:
Of course you’re right about the implications of the duplication of information. But not wanting to be finicky about it, I still think that a PICO board user should not have to ‘compose’ the information he/she needs from datasheets of individual components of the board. The statement ‘can be viewed as a breakout board’ for me – not native English – does not follow from the sentences in the pico datasheet saying e.g. ‘Raspberry Pi Pico is a microcontroller board based on the Raspberry Pi RP2040 microcontroller chip’. and ‘Raspberry Pi Pico has been designed to be a low cost yet flexible development platform for RP2040’ 

The power consumption data on page 11 of the pico datasheet is (thus) also related to the board, and not one chip on it … Fundamentally, there are problems mixing up BOARD level interface specs with CHIP level interface specs – unless there is a galvanic connection between the two. So it can easily be solved by stating in the PICO datasheet that GPIO pins of the board are those of the chip (‘breakout board’) and that for voltage/current specs of those particular pins the user should download/obtain yet another document and look at the Electrical/Mechanical section. Such reference can hardly become out of sync. Just tryin’to be helpful on behalf of those users that want merely the essentials allowing them to reliably use the board. For which, by the way, with e.g. the Python SDK and its support in hardware and software you did an awfully good job J

lurch, 2021-03-04:
I also don't want to get finicky about it (and suggestions are always appreciated :+1: ), but if the information a user is looking for isn't in the Pico datasheet, surely the next place they'd look would be the RP2040 datasheet? :shrug:  Or am I missing something here?

I'm only a "software person" and not a "hardware person", but even to me it seems that there are so few components on the Raspberry Pi Pico that it seems fairly obvious that the pins on the Pico board must be directly connected to the RP2040 chip?

PeterPotgieser, 2021-03-04:
The exchange starts to go in a direction I tried to avoid; the dimension now added is about presumed user subject matter / inside knowledge. For starters, I really do think that a supplier should provide information on the interface level of the item he/she delivers. So it is either BOARD level, or CHIP level. (If you purchase a USB3 add-on board for a PC to connect some external device, you rely on the specs of the interface of the board (i.e. the USB3 sockets and the edge connector), and not of the chip(s) on it. And, by the way, the PCI-Express slot that the USB3 add-on board needs to be inserted into also has its interface specified at the connector and not at the (one or more ?) chips driving it.) Unless, of course, the supplier takes a shortcut and for BOARD level interface descriptions refers the user to (elsewhere existing) CHIP level interface descriptions (e.g. to avoid duplication of information as you mentioned). I think it is a matter of ‘service level’ by the supplier towards its customers:  make it easy for the customers to unequivocally and simply find the information they need to reliably use the board without having to identify themselves necessary fragments in other documents. A matter of ‘layered’ principle.

 

The other thing is the ‘surely’ and ‘fairly obvious’. 21st century technology allows for the use of techniques that are almost invisible to the naked eye. For instance, on a Raspberry PI Zero the RAM chip seems to be soldered on top of the CPU. You can’t see that. So, on a PICO, why couldn’t the visible chip (the SoC ?) be soldered on top of something below it that contains e.g. RAM or GPIO buffer drivers ? The ordinary user has no means to know that and for the use of the BOARD it should be irrelevant. Does the user, wanting to use the PICO board reliably (i.e. without a risk of too much power dissipation or current overloads at BOARD level) really need to investigate or should the supplier (the only party with the proper knowledge) prevent this research and help its user by simply cancelling all doubts beforehand by adding one or two lines (as suggested) in a BOARD level document ?  I do fully understand that for the developers of the BOARD there are things so obvious that they are not automatically documented (initially) for those not having the detailed subject matter knowledge, but I would hope these ambiguities / omissions are corrected in subsequent versions. 

Thanks for your efforts …. For me this closes the exchange …. The decision is of course yours.

daveythacher, 2021-03-06:
I agree with your statements. The RP2040 appears to be designed to be very similar to BCM283x in design. It is documented in a similar manner. Most documentation is casual. There is not a lot of the detailed specs you would normally see in datasheets. Part of this is due to the nature of this GPIO design. It mostly expects external drivers. It works with weak drivers which may perform well at high speed. (Most IO drivers are weak.) This driver model itself acts a lot like FPGA drivers. Many MCU drivers have some of these properties and some FPGA drivers have more.

The RP2040 is not going to compete against a AVR MCU. The AVR uses stronger/more bank internally for cost reduction purposes. The RP2040 uses weaker/less for cost reduction purposes. Again these specs are similar to the Pi under all versions if I am not mistaken. Most of the specs here were also lacking.

Granted I will say that was more permissive due to being MPU. I mentioned a list of other things not covered in issue 37, but it was removed. However I also stated that most if not all of those concerns are mitigated with external buffer drivers. This will add cost and space to the design. However the RP2040 drivers may be able to work better for some applications in this current configuration.

The RP2040 is experimental MCU. Its is very software centric like many ARM, MPU and Atmel implementations, baring the exceptions that do also exist. I will say this again the RP2040 has full IPC and PIO as its leading features. It has a few unique traits in other areas, but those require custom implementation at board layout.

The power consumption test was very casual indeed. Hot vs cold, overclocking, random code test, no internal test parameters, etc. However given the sleep currents this will not likely be going head to head with power optimized MCUs. Therefore an argument can be made against specifying some of this.

The GPIO topology here follows FPGA in design. It attempts to move all IO current to a single power pin. However ground is shared.  Given a single GPIO drive bank capped to 50mA this will reduce the overall IO power consumption significantly. All the remaining power is lower voltage pins for CPU, PIO, SPI, etc. These will scale with frequency and likely make it nearly impossible to exceed the power limit.

The Pi used two GPIO banks but only exposed one without compute module. The Pi had the same 50mA limit per bank. If you damage the bank driver you screwed, but pin drivers may continue to operate independently if damaged.

Raspberry Pi Foundation's original purpose was education. It tried to make things more approachable yet not hinder the learning. From this I would not be surprised to see a few levels of documentation. Many tutorials leave this out. It is very possible to overload the Pi and RP2040.

Overall documenting the PICO as a chip of its own, given its SMD nature would not be unreasonable. Making an APP note or shorten document would also not be unreasonable. I am not sure how many people would even look for it. I never knew the exact limit of the Pi and never had a problem. Granted there is a few reasons for that.

Please remember that these "issues" serve as documentation. Please know what is and is not off topic.

PeterPotgieser, 2021-03-06:
“Raspberry Pi Foundation's original purpose was education” … and I acknowledged ‘they’ did a very good job. But …. that is even more reason not to presume anything about the subject matter knowledge of (potential) users … and hence reason to provide adequate information to them.  I can imagine designers / developers overlook to include such basic details in usage guides, but then one would hope they use feedback ‘from the field’ to help fill in the white spots. I rest the case ….