raspberrypi/pico-feedback #113: Interrupt handling documentation issues

MichaelBell, 2021-03-05:
In the C/C++ SDK docs section `4.1.13.3.11. irq_set_priority`, the priority levels are described backwards.  The ARM docs and the RP2040 datasheet are clear that the lower value corresponds to a higher priority, but the C/C++ SDK docs incorrectly say "Hardware  priorities  range  from  0  (lowest)  to  255  (highest)"

Also, it would be nice if section `2.4.5.1 About the NVIC` of the RP2040 datasheet explicitly said that higher priority interrupts can interrupt lower priority interrupt handlers.  I wasn't sure whether the priorities worked like this or just determined which interrupt would be serviced first, but I'm told on ARM processors higher priority interrupts always preempt lower priority interrupts.

Wren6991, 2021-03-05:
> In the C/C++ SDK docs section 4.1.13.3.11. irq_set_priority, the priority levels are described backwards

Thanks for spotting this, the doxygen is wrong there. I'll raise a PR for it (it's generated from the SDK header)

> Also, it would be nice if section 2.4.5.1 About the NVIC of the RP2040 datasheet explicitly said that higher priority interrupts can interrupt lower priority interrupt handlers. I wasn't sure [...]

I had a quick scroll through and I was surprised to see that it doesn't mention this. Especially surprising as that's what the N in NVIC stands for! (Nested interrupts). 2.4 is excerpted from the M0+ technical reference manual:

https://developer.arm.com/documentation/ddi0484/latest

Which itself defers to the ARMv6-M architecture reference manual for all the details on exceptions:

https://developer.arm.com/documentation/ddi0419/latest/

But that is a fairly hefty document so it's not helpful to just link that in. I'll add some hints to our section 2.4.

lurch, 2021-03-05:
> Especially surprising as that's what the N in NVIC stands for! (Nested interrupts).

My brain keeps forgetting that that acronym doesn't stand for Non Volatile... :laughing:

Wren6991, 2021-03-10:
I dropped some pointers and details in the first few places I thought to look:

[image]

[image]

[image]

MichaelBell, 2021-03-10:
Looks great to me!

closed 2021-03-11 by aallan

b-zee, 2022-04-28:
I came across this issue while looking at interrupt preemption based on the IRQ number (equal dynamic level). I submitted that issue here. https://github.com/raspberrypi/pico-feedback/issues/233

I think the documentation is vague, as it appears there is a difference between the priority that determines nested preemption and the order in which pending exceptions are activated...