raspberrypi/pico-feedback #114: Documentation missing for the pico_unique_board_id_t (and other?) struct members

guruthree, 2021-03-05:
In the C SDK, `pico_get_unique_board_id()` uses the struct `pico_unique_board_id_t`. However, there is nothing there to say what the members of the struct are. Section 4.2.13.1 "Data Structures" looks like the place this information should be.

I also don't think there are members listed for any other structs, e.g. Section 4.2.15.1 "Data Structures" for `datetime_t`.

lurch, 2021-03-05:
Thanks for the report, I'll take a look at this next week (on the back of my success with https://github.com/raspberrypi/pico-sdk/pull/212 ).

lurch, 2021-03-30:
Ahhh, just looked at this again and it turns out to be a bit trickier than I first thought...

I initially thought it was just due to missing Doxygen in the relevant headers in https://github.com/raspberrypi/pico-sdk , but those structs *do* have Doxygen as they appear at https://raspberrypi.github.io/pico-sdk-doxygen/structpico__unique__board__id__t.html and https://raspberrypi.github.io/pico-sdk-doxygen/structdatetime__t.html . It's the process that creates the C SDK book from the XML exported from Doxygen that needs updating, and @aallan knows how much I enjoy that... :laughing:

aallan, 2021-05-25:
@lurch Did this get done or is it on backlog?

lurch, 2021-05-25:
It's still in my backlog...

lurch, 2022-12-02:
Closing as invalid, as the SDK API docs are now only at https://raspberrypi.github.io/pico-sdk-doxygen/ and no longer included in the C-SDK PDF.

closed 2022-12-02 by lurch