raspberrypi/pico-feedback #115: How to link to sub-pages of Pico website?

todbot, 2021-03-05:
It is not possible to link to sub-pages of the site at https://www.raspberrypi.org/documentation/rp2040/getting-started/

For instance, I want to link to the "Resetting Flash memory" area. But I can only give the above `getting-started` URL with the attendant paragraph: 

> "Scroll down to the four tags area ("RP2040 On Board",etc), then click on "About Raspberry Pi Pico". Scroll down more to the "Utilities" section, then to the "Resetting Flash memory" section."  

One can't even say "Go to the `getting-started` page and Cmd-F for `resetting flash`" because the sub-pages aren't loaded or visible to the browser.

Perhaps it's possible to add anchors to the various sub-pages and sections and still maintain the SPA-aspect of the site?

aallan, 2021-03-05:
@mudge Any thoughts?

aallan, 2021-03-08:
Hey @todbot, @mudge is looking into this and it looks like it's fixable. Will let you know when and if!

mudge, 2021-03-08:
Hi @todbot,

We've just deployed an update to the site so that it is possible to link directly to a tab, e.g. https://www.raspberrypi.org/documentation/rp2040/getting-started/#board-specifications

We don't currently have IDs or anchors for sub-sections but hopefully this means that you can link to a specific tab and `Cmd-F` should work as expected.

Clicking between the tabs will update your browser's URL with the appropriate tab in the fragment identifier but here they are for posterity:

* https://www.raspberrypi.org/documentation/rp2040/getting-started/#rp2040-boards
* https://www.raspberrypi.org/documentation/rp2040/getting-started/#board-specifications
* https://www.raspberrypi.org/documentation/rp2040/getting-started/#getting-started-with-micropython
* https://www.raspberrypi.org/documentation/rp2040/getting-started/#getting-started-with-c

Also note that these URLs should also work when viewing the site with a narrow screen (e.g. on mobile) as the tabs are replaced with a Table of Contents list with regular links.

lurch, 2021-03-08:
I've also verified that these work with our short-links (as used by e.g. https://raspberrypi.github.io/pico-sdk-doxygen/ ) e.g. https://rptl.io/rp2040-get-started#board-specifications

closed 2021-03-08 by aallan

todbot, 2021-03-08:
Looks and works great, thanks!