raspberrypi/pico-feedback #116: Q: Atomic Register Access Upon Certain Registers

soundpaint, 2021-03-06:
The effect of atomic register access as defined in RP2040 datasheet, Sect. 2.1.2 seems unclear to me when applied onto some specific registers.

Example 1: Atomic XOR on write (e.g. via hw_xor_bits()) applied onto PIO Register TXF0.  This register is, as far as I understand, a write-only register, since it pushes a new data word into the FIFO.  Applying XOR however requires knowledge about the bits of a previous value, but there is no previous value when pushing a _new_ value into the FIFO.  My guess is, that the XOR is applied against an assumed 0x0 value, i.e. resulting in the same effect as directly writing to the register without applying XOR.  Is this correct?

Example 2: Atomic bit mask set and clear (e.g. via hw_set_bits() and hw_clear_bits) applied onto PIO Register INSTR_MEM0.  The memory words are write-only.  If atomic bit mask set and clear nevertheless worked, this would mean that you could non-destructively update the delay/side-set of an instruction without knowing what instruction is currently stored: Just clear/set the delay/side-set bits of an instruction with a mask that keeps all other bits of the instruction unmodified.  Does this really work?  My guess is no...

Also, the atomic register access is specified for _write_ operations only.  What will be the effect when _reading_ from an address that is provided e.g. for atomic XOR write access?

aallan, 2021-05-25:
@Wren6991 Thoughts?

kilograham, 2021-05-25:
Note i also just realized (from watching a live stream) that this can be confusing in other cases (although perhaps there is a simple "rule of thumb"). The case in question was `hw_clear_bits` of a WC register ... this actually makes sense (though is unnecessary), however it makes me wonder what the other variants do here