raspberrypi/pico-feedback #118: Using Pico on Windows 7/8 is easier than documented

ndabas, 2021-03-06:
In the _Getting Started_ guide, section 9.2 Building on MS Windows, there is a warning note that says:

> Using Raspberry Pi Pico with Windows 7 or 8 is not officially supported but can be made to work.

This links to a forum discussion with solutions that I would describe as ad-hoc at best. The basic issue at hand is that Windows 7 and 8 do not auto-install a serial port driver for the Pico's CDC interface -- this is easily fixable using Zadig. Zadig is mentioned elsewhere in the guide to get Picoprobe running as well, so it makes sense to use the same tool for this as well. Something like:

If you need to connect to the USB serial interface on your board using Windows 7 or 8, you will need to tell Windows to use the appropriate built-in driver. (Windows 10 installs the correct driver automatically.)

We will use Zadig (https://zadig.akeo.ie/) for this.

Download and run Zadig.

Select Board CDC (Interface 0) from the drop-down box. Select USB Serial (CDC) as the driver.

[image]

Then click Install Driver. A serial port will be assigned when the installation is complete.

Saultes45, 2021-04-16:
After the uf2 downloading it didn't even show up as "Board CDC" but it was present in the device manager. I had to unplug/replug 4 times to see it as "Board CDC". 
Worked well after.

lurch, 2021-04-16:
@Saultes45 Was your experience similar to the one described here? https://github.com/raspberrypi/pico-sdk/issues/257

Seems very odd that unplugging and replugging 4 times magically fixed it, but we don't officially support older Windows versions so meh :shrug: :wink:

Saultes45, 2021-04-16:
> @Saultes45 Was your experience similar to the one described here? raspberrypi/pico-sdk#257

Yes, that was the same.

I confirm that strange behaviour. I made the mistake of saving a py file and crashed the pick. This I had to redo the whole procedure from the bootloader drag-drop and the driver installation via Zadig. Between the 2, I had to unplug and replug it at least 3 times.

I am curious why that happened, though.

Regards,

Nathanaël Esnault

ndabas, 2021-04-16:
> After the uf2 downloading it didn't even show up as "Board CDC" but it was present in the device manager.

@Saultes45 Do you mean that the serial port interface was present in Device Manager (without the driver installed), or was it just the USB mass storage device? It seems you were using MicroPython here.

- If the serial port was showing up in Device Manager but not in Zadig: could be some condition where Zadig did not display the USB CDC interface for some reason.
- If the serial port did not show up in Device Manager at all - it could be something in the MicroPython code (either the interpreter itself or your .py file).

Saultes45, 2021-04-16:
>
> It seems you were using MicroPython here.

That is correct I am using a vanilla version of Thonny with interpreter
"MicroPython (Raspberry Pi Pico)" which I assume is the latest one.

Do you mean that the serial port interface was present in Device Manager
> (without the driver installed), or was it just the USB mass storage device?

It was present in the device manager as "Unknown device" when I plugged it
back again in the computer. I had a warning notification on the bottom
right explorer toolbar from Windows saying that the USB device was not
recognised.

>    - If the serial port was showing up in Device Manager but not in
>    Zadig: could be some condition where Zadig did not display the USB CDC
>    interface for some reason.
>
> That is possible, even if you see select the option to list all devices.

>    - If the serial port did not show up in Device Manager at all - it
>    could be something in the MicroPython code (either the interpreter itself
>    or your .py file).
>
> Usually, I assume I f**k up somewhere ^^. The 2 problems I had were
forgetting to import the "Time" module but calling it and saving a python
file on the pico but forgetting the .py extension and still trying to run
it.

BrianG61UK, 2021-05-04:
I don't want to use zadig since I find it totally unintuitive and it has driven me to completely wiping my C: drive when trying to undo it's effects for RTL2382U dongles.
Please can someone provide a normal .inf file driver for Windows 7 64 bit that actually works for this.

aallan, 2021-05-05:
I'm sorry, but we do not intend to officially support Windows 7 as Microsoft itself discontinued support for the operating system in January last year.

closed 2021-05-05 by aallan

BrianG61UK, 2021-05-06:
I see. But you can still pay for Windows 7 support you know.

JamesH65, 2021-05-06:
None of the money that MS get for support for Win 7 gets to Raspberry Pi.

Tomwhiting1987, 2021-11-01:
I have managed to stop my Pico from showing up as a mounted drive in BOOT mode. This happened after I changed the drivers using Zadig to the CDC SErial ones on RP0 and FP1 Interfaces. Any ideas how I can change these back so that the Pico shows up as a drive again?

aallan, 2021-11-01:
I'm sorry, but we do not intend to offer any official support for Windows 7 or 8.

IdoHic, 2022-04-03:
thank you very much, you were very helpful

matsobdev, 2023-03-22:
It is possible to export ini driver files from Zadig instead of installing it.

larypie, 2023-04-06:
Here is a way to install the Pico serial driver on Win 7...

Download the **pico-serial.inf** file from here [https://github.com/Melvinsajith/raspberrry_Pi_Pico_windows_7_8_xp](url)

Open Device manager and navigate to  the Pico deice - on my computer it showed up as Board CDC in "Others".  

Select : Properties / Driver / Update Driver /Browse my Computer / Let me pick / Have Disk / Select : Pico-Serial.inf

Other options are probably available, but that worked for me.

CODE-REaD, 2023-07-17:
The pico-serial.inf file referred to here worked for me, after I discovered that I had downloaded it improperly.  If your downloaded .inf file is a text file consisting of **_one line_**, you have not downloaded it correctly.  It should appear as a typical Windows .inf file, whose first "stanza" starts with "[Version]".

Once I correctly downloaded, my _Windows 7 Professional 64 bit  6.1.7601 Service Pack 1 Build 7601_ OS accepted the pico-serial.inf as the driver for my _Board CDC_ device after confronting me with a red-framed popup alerting me to the fact that this device driver is not signed by Microsoft.

I did not need to install or use Zadig to make this happen.  I can tell I am connected to my Pico using Thonny, where my Pico shows up in its menu of interpreters (bottom right of main screen).

Briefly, the whole process for me was,
1. Obtain correct uf2 file for my pico
2. Connect my Pico to my Windows 7 computer via USB cable
3. Drag and drop uf2 file to my pico's _RPI-RP2_ directory as described **[here](https://www.makeuseof.com/getting-started-micropython-raspberry-pi-pico/)**
4. My pico shows up in Windows Device Manager as _Board CDC_ under _Unknown device_
5. Using Device Manager, I right-click _Board CDC_ and then _Update Driver Software_ and select pico-serial.inf as described by @larypie above
6. Windows warns me that this driver is not signed my Microsoft
7. I instruct Windows to install anyway
8. My pico is listed by Windows Device Manager under _(Ports COM & LPT)_ as _Pi Pico Serial Port (COM25)_
9. Thonny recognizes my Pico as _MicroPython (Raspberry Pi Pico) - Board CDC @ COM25_.

carypt, 2023-09-17:
because the above link to the .inf-file directs to an empty issue site, search the .inf-driver-file here : https://github.com/Melvinsajith

lurch, 2023-09-17:
> because the above link to the .inf-file directs to an empty issue site, search the .inf-driver-file here : https://github.com/Melvinsajith

Thanks, the link in the aforementioned comment was formatted incorrectly, so I've now edited it so that it should work properly.

ndabas, 2023-09-18:
Just pointing out for readers of this thread that they should not attempt any of this on Windows 10/11 or later, this is really only if you must use a Pico/RP2040 on Windows 7/8. Even if so, please use [Zadig](https://zadig.akeo.ie/) rather than the inf file directly.