raspberrypi/pico-feedback #119: 'Drag and drop MicroPython' section has inconsistent download links

scruss, 2021-03-06:
Page https://www.raspberrypi.org/documentation/rp2040/getting-started/ has two links to download a MicroPython UF2:

* in the body text: the link "downloadable UF2" goes to https://micropython.org/download/rp2-pico/
* in the "Download UF2 →" button, the link goes to https://www.raspberrypi.org/documentation/rp2040/getting-started/static/f70cc2e37832cde5a107f6f2af06b4bc/rp2-pico-20210205-unstable-v1.14-8-g1f800cac3.uf2

Shouldn't the button at least point to the micropython.org downloads?

Thonny (as far as I can tell) pulls in https://micropython.org/resources/firmware/rp2-pico-20210202-v1.14.uf2 - the only non-unstable UF2 version

lurch, 2021-03-07:
There was a bug in the stable v1.14 release of MicroPython ( https://github.com/micropython/micropython/issues/6837 ), which is why we explicitly link to that particular unstable version (which contains the fix). I believe the plan is that after [MicroPython v1.15](https://github.com/micropython/micropython/issues/6832) is released, we'll then revert back to linking directly to the micropython.org site.

scruss, 2021-03-07:
but you *do* link directly to micropython.org already on that page

aallan, 2021-03-07:
> Thonny (as far as I can tell) pulls in https://micropython.org/resources/firmware/rp2-pico-20210202-v1.14.uf2 - the only non-unstable UF2 version

The version Thonny installs is bugged (paging @aivarannamaa), see micropython/micropython#6837, which is why the Download UF2 button links to an unstable build.

> but you do link directly to micropython.org already on that page

...and I'll fix that to point to the unstable build.

Both these links will point to v1.15 when it releases.

closed 2021-03-07 by aallan

aivarannamaa, 2021-03-07:
@aallan, thanks for the ping! I didn't know about 20210205 version. 

https://raw.githubusercontent.com/thonny/thonny/master/data/rpi-pico-firmware.json is now updated and Thonny users will get the version with `array` from now on. Upgrading Thonny is not required.

Please ping me again when there is an important update which is not mentioned under https://github.com/micropython/micropython/releases (I'm following it).