raspberrypi/pico-feedback #121: WS2812 Diver code run through in raspberry-pi-pico-c-sdk doc.

Bleep42, 2021-03-09:
In raspberry-pi-pico-c-sdk doc, Section 3.2.2 Pg 37 on.
The Times specified in the code and on page 37, are T1=2, T2=5, T3=3 however the timing diagrams further down Fig 4 to 7 imply timings of T1=4 T2=3 T3=3 assuming I am reading them correctly. Also Fig 6 mentions T1 twice, I think the second mention should be T2.
Assuming I understand how the code works, I believe the actual timing values should be T1=2, T2=2, T3=3 which I believe should give 358ns for a '0' bit, 716ns for a '1' bit and a minimum inter bit spacing of 537ns. My understanding is that the '1' bit should be roughly twice the length of the '0' where as in the example it is over 3 times?  (2clk + 5clk) for a '1' or just 2clk for a'0'
Though I understand that the original timings will work fine, (they do) it doesn't help to make how the code works very clear, especially when looking at the timing specifications for the WS2812, I also understand that the specification timings  can vary quite a bit depending on where you look, or 2,4,2 potentially if a WS2812B.

henrygab, 2021-03-12:
First, I am impressed with the analysis and learnings shown in this thread.

As you may know, [FastLED](https://github.com/FastLED/FastLED) is a well-known, heavily-tested and optimized library for controlling these LEDs ... some would say they are the "gold standard".  Thus, for real-world use, I would point to the [FastLED WS2812 timings](https://github.com/FastLED/FastLED/blob/1fb0d7b2afd87695eb5f3c4af27c286b5e9f5fe0/src/chipsets.h#L576-L578) as the timings I would recommend for use in examples.

As to the WS2812 vs WS2812B, they have the [same timing](http://rgb-123.com/ws2812b-vs-ws2811/).

psitech, 2021-03-12:
Right, FastLED is probably where the initial Pico SDK "2-5-3" timing originated from:
> // WS2812 - 250ns, 625ns, 375ns
template <uint8_t DATA_PIN, EOrder RGB_ORDER = RGB>
class WS2812Controller800Khz : public ClocklessController<DATA_PIN, C_NS(250), C_NS(625), C_NS(375), RGB_ORDER> {};

And although that timing seems to work in general, it's formally violating the LED specs. But the again, the LED specs are also ambiguous. 
I do have a preference for the 1/3-2/3 timing, since you can achieve that timing with smallest PIO code:

```
@rp2.asm_pio(sideset_init=rp2.PIO.OUT_LOW, out_shiftdir=rp2.PIO.SHIFT_LEFT, autopull=True, pull_thresh=24)
def ws2812():
    wrap_target()
    label("bitloop")
    out(x, 1)               .side(0)
    jmp(not_x, "do_zero")   .side(1)
    jmp("bitloop")          .side(1)
    label("do_zero")
    nop()                   .side(0)
    wrap()

sm = rp2.StateMachine(0, ws2812, freq=2_400_000, sideset_base=Pin(PIN_NUM))
```

But the final decision is not up to me :-)

aallan, 2021-05-25:
I'm somewhat unclear as to what the conclusions from this thread actually were? Is there still a documentation issue?

Bleep42, 2021-05-25:
Nope. nothing has changed.
As I said in the very first post.
"In raspberry-pi-pico-c-sdk doc, Section 3.2.2 Pg 37 onwards
The times specified in the code and on page 37, are T1=2, T2=5, T3=3 however the timing diagrams further down Fig 4 to 7 imply timings of T1=4 T2=3 T3=3 assuming I am reading them correctly. Also Fig 6 mentions T1 twice, I think the second mention should be T2." Personally I think the timing diagrams are making things confusing, by being different to the code. The remainder of the comment is open to debate.
Regards, Kevin.

psitech, 2021-05-25:
Yes, Kevin is right. Figure 3, page 35, [C SDK, rev 1.4.1] does not match the actual timing in the example code on page 36.
Then the thread went into the details of what the timing should be, mea culpa.
So the documentation is not 100% correct.

Bleep42, 2021-05-28:
Hi, 
Actually the Fig. 3 page 36 is basically ok, it's just an approximate, this is the shape of the waveform, type diagram.
However Fig. 4 to 7, Page 38 to 40 inclusive all show fairly specific timing information, T1=4, T2=3, T3=3. This does not agree with the code, which specifies that T1=2, T2=5, T3=2.
```
10 .define public T1 2
11 .define public T2 5
12 .define public T3 3
```
Regards, Kevin.

lurch, 2021-05-28:
Also, the caption for Figure 4. talks about T1 and T2, whereas the diagram itself shows T3 and T1.

Bleep42, 2021-05-28:
Hi Andrew,
You are quite right, I hadn't noticed that. Infact both the T's are wrong, should probably be something similar to this "The statemachine drives the line low for time T3 as it shifts out one data bit from the OSR, and then high for time T1 whilst branching on the value of the bit."?
Similarly as I have already pointed out in Fig. 6 T1 is mentioned twice, I suspect the second mention should infact be T2.
Regards, Kevin.

lurch, 2021-05-28:
I'd need to read and understand the whole section to determine whether it was the caption that was wrong or the image that was wrong :wink:   And unfortunately I'm far too busy with other stuff to have time for that.

Bleep42, 2021-05-29:
I have, it looks correct to me.  ;-)
Regards, Kevin.