raspberrypi/pico-feedback #125: Recursive mutex

Wiz-IO, 2021-03-14:
Please support **recursive mutex**
mutex_enter_blocking_recursive()
mutex_try_enter_recursive()
mutex_exit_recursive()

lurch, 2021-03-14:
ping @kilograham

Wiz-IO, 2021-03-19:
just for info ... look this
https://github.com/Wiz-IO/LIB/blob/master/pico/pico_mutex/mutex.c#L82
new is only one counter in **mutex_t**
https://github.com/Wiz-IO/LIB/blob/master/pico/pico_mutex/mutex.h#L35

results with <sys/lock.h>
[no locks](https://raw.githubusercontent.com/Wiz-IO/LIB/master/pico/no-lock.jpg)
[with locks](https://raw.githubusercontent.com/Wiz-IO/LIB/master/pico/lock.jpg)

kilograham, 2021-03-19:
> results with <sys/lock.h>
> no locks
> with locks

Not 100% sure what you are trying to indicate here... I know what a mutex does :-) ... FYI printf in the SDK is already mutex-ed for you by default.

Wiz-IO, 2021-03-19:
recursive mutex is used for "future" ..... <sys/lock.h> as baremetal and/or FreeRTOS to protect the stdio and file system

Wiz-IO, 2021-03-19:
example: **unistd** and **FILE** in **Arduino (C++)**
RAM Disk + FLASH Disk + SD Card
posix? :)
[image]