raspberrypi/pico-feedback #129: Documentation: RP2040 Datasheet - IC_SDA_HOLD Register

fivdi, 2021-03-22:
The following sentence is the last sentence on page 511 of release 1.3.1 of the RP2040 Datasheet and is part of the description of the IC_SDA_HOLD Register.

>The value programmed in IC_SDA_TX_HOLD must be greater than the minimum hold time in each mode one cycle in master mode, seven cycles in slave mode for the value to be implemented.

There is something wrong with the punctuation and/or structure of this sentence making it difficult to understand.

lurch, 2021-03-22:
The part you're referring to is documentation automatically extracted from the IPXACT file that comes with the Synopsys I2C controller. As it notes at the start of section 4.3:
> Synopsys Documentation
Synopsys Proprietary. Used with permission.

I wonder if it _might_ have lost some brackets? :shrug:  "The value programmed in IC_SDA_TX_HOLD must be greater than the minimum hold time in each mode (one cycle in master mode, seven cycles in slave mode) for the value to be implemented."

I'll double-check tomorrow to check that it's not an error in the document-processing that we do on "our side".

fivdi, 2021-03-23:
> I wonder if it might have lost some brackets? shrug "The value programmed in IC_SDA_TX_HOLD must be greater than the minimum hold time in each mode (one cycle in master mode, seven cycles in slave mode) for the value to be implemented."

Yes, those brackets would fix it.

lurch, 2021-04-09:
I checked last week, and even the Synopsys databook says "The value programmed in IC_SDA_TX_HOLD must be greater than the minimum hold time in each mode one cycle in master mode, seven cycles in slave mode for the value to be implemented" :confused: 

But one of the hardware engineers agrees that the sentence makes more sense with the brackets inserted, so that's what will be appearing in the next version of our PDF documentation :smiley:

closed 2021-04-09 by lurch