raspberrypi/pico-feedback #130: Q: How to Externally Reset a PIO's GPIO Output Flip-Flops?

soundpaint, 2021-03-22:
RP2040 datasheet Sect. 3.5.6.1. ("Output Priority"), Fig. 49 depicts per-GPIO priority select of write masks from each state machine.  For each GPIO pin, the figure depicts 2 flip-flops that are set / reset via the priority mask.

Image that some PIO program has been executed and left some flip-flop in state 1.  Now, assume that the SMs are disabled, a new PIO program is loaded, and all SMs are reset and started in order to execute the new program.
As far as I understand, the flip-flop will now still be in state 1 even after resetting all SMs.  Is this correct?  Then, the only way to reset the flip-flop is the new PIO program itself actively initialising all relevant output flip-flops?  Or is there a way "outside" of a PIO program to reset these flip-flops via the registers (just like resetting an SM itself) to have a defined startup status of these flip-flops for a PIO program?

(Maybe a nice extension for a future version of the RP2040 would be to make PIO register DBG_PADOUT writable for exactly this job, though the name of this register does not perfectly fit...?)

lurch, 2021-03-23:
>  Or is there a way "outside" of a PIO program to reset these flip-flops via the registers

`CTRL.SM_ENABLE` says "When disabled, a state machine will cease executing instructions, except those written directly to SMx_INSTR by the system", so I guess disabling the state-machine and then writing to a state-machine's `INSTR` register _might_ be one way of doing that?

Wren6991, 2021-03-23:
There are some convenience functions for this. See for example [pio_sm_set_pins_with_mask()](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pio.html#ga963167ca42f67f79ec5de6b47d6027d5) or [pio_sm_set_pindirs_with_mask()](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pio.html#ga8269f1ef4cb1e4b3b957d5f1fa6fc623), which pokes those registers by directly executing SET instructions on a victim state machine.

lurch, 2021-03-23:
@Wren6991 I know very little about PIO, but I guess in theory you could have a `pio_sm_set_all_pins(PIO pio, uint sm, bool pinval)` which does things with a single `MOV` rather than using `SET` on 5 pins at a time?

Wren6991, 2021-03-26:
Yes, you could just execute a `mov pins, null` or `mov pins, ~null` for that case. The SDK function is trying to solve the general problem, and for a one-time setup like this the performance is not critical.

closed 2021-06-04 by aallan