raspberrypi/pico-feedback #131: sideset opt in datasheet and SDK book

benevpi, 2021-03-29:
```
side_set <count> (opt) (pindirs) 

If this directive is present, <count> indicates the number of side set bits to be
used. Additionally opt may be specified to indicate that a side <value> is
optional for instructions (not using this requires stealing an extra bit - in
addition to the <count> bits - from those available for the instruction delay).
Finally, pindirs may be specified to indicate that the side set values should be
applied to the PINDIRs and not the PINs. This directive is only valid within a
program before the first instruction
```

I *think* the 'not'  in "not using this requires stealing an extra bit" should be 'note'. If I understand it correctly (and I may not), if you use opt, this uses one of the sideset/count bits in each instruction so there is one fewer bit available. 

I double-checked the source of PIOASM, to see if my understanding for this is correct and:

```
int sideset_bits_including_opt; // specified side set bits + 1 if we need presence flag
```

So, it looks like it.

This text is in both the RP2040 datasheet (section 3.3.1) and the C SDK book (3.3.2).

lurch, 2021-03-29:
Duplicate of #109 ?

benevpi, 2021-03-30:
Yep, duplicate of 109: https://github.com/raspberrypi/pico-feedback/issues/109

closed 2021-03-30 by benevpi