raspberrypi/pico-feedback #132: pio_encode_xxxxx functions

benevpi, 2021-03-29:
Not sure if  I'm just looking in the wrong place, but the PIO_encode_xxx functions such at 
pio_encode_pull and pio_encode_out (see full list in https://github.com/raspberrypi/pico-sdk/blob/fc10a97c386f65c1a44c68684fe52a56aaf50df0/src/rp2_common/hardware_pio/include/hardware/pio_instructions.h) don't seem to be documented anywhere. I could be looking in the wrong place, but I couldn't find them in the RP2040 data sheet, the C SDK book or the micro-site (https://raspberrypi.github.io/pico-sdk-doxygen/).

lurch, 2021-03-29:
Kinda related to this, I've got a local change (which I've not yet tidied up and PRed) which adds the "missing" 
```
pio_encode_jmp_not_x
pio_encode_jmp_x_dec
pio_encode_jmp_not_y
pio_encode_jmp_y_dec
pio_encode_jmp_x_ne_y
pio_encode_jmp_pin
pio_encode_jmp_not_osre
pio_encode_irq_set_wait
```
encoding functions (which I haven't added Doxygen to either) to the `pio_instructions.h` file.

The pio_encode_xxx functions don't currently appear in the RP2040 datasheet or C SDK book because they don't have any Doxygen comments attached to them. My assumption was that this was deliberate, as these are supposed to be "internal use only" functions? But that's ultimately a decision for @Wren6991 or @kilograham to make...

kilograham, 2021-03-29:
nope, not deliberate... just didn't get to it b4 launch

aallan, 2021-05-25:
@kilograham Are these now documented in the Doxygen (and hence the C SDK book)..?

lurch, 2021-05-25:
Nope, not yet https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_pio/include/hardware/pio_instructions.h#L81

lurch, 2021-10-22:
> adding this doc to SDK1.3.0

...in https://github.com/raspberrypi/pico-sdk/pull/624