raspberrypi/pico-feedback #134: 8.2. Working in Visual Studio Code -- CMake configuration

michaelstoops, 2021-04-06:
I tried to follow the instructions to use VS Code for the `test` project as described in chapter 8. I found that the command-line instructions seemed to work right, which is great. I opened the folder in VS Code as described in _7.2. Loading a Project_, but I didn't get the popup for selecting a tool kit, nor did the buttons appear on the bottom bar.

It looks like the missing step is to open the Command Palette (⇧⌘P) and run **CMake: Configure CMake**.

This article was how I found my solution: https://code.visualstudio.com/docs/cpp/cmake-linux#_select-a-kit), although their recommended command: "CMake: Select a Kit" doesn't exist in my UI.

I don't know whether this is an isolated issue or not, but I figured I'd share. :)

aallan, 2021-04-06:
Not something we've come across to be honest. I'll poke around and see if I can replicate.

lurch, 2021-04-06:
I _think_ I saw this myself once, but I was busy with other stuff at the time, so I just assumed I'd done something wrong and stuck with building from the command-line. I'll assign to myself and investigate later...

aallan, 2021-05-25:
@lurch Did you manage to take a look, or is this still backlog?

lurch, 2021-05-25:
Still in backlog.

lurch, 2025-07-09:
Closing this, as this was referring to the old way of doing things, before we had https://github.com/raspberrypi/pico-vscode

closed 2025-07-09 by lurch