raspberrypi/pico-feedback #137: Documentation: RP2040 Datasheet - Section 4.3.15.4. Operation of Interrupt Registers

fivdi, 2021-04-09:
In release 1.4 of the RP2040 Datasheet, section 4.3.15.4. "Operation of Interrupt Registers" is a subsection of section 4.3.15. "DMA Controller Interface".

This is strange as the "Operation of Interrupt Registers" doesn't have a much to do with the "DMA Controller Interface".

Perhaps it would be better if section 4.3.15.4. "Operation of Interrupt Registers" was a subsection of section 4.3 "I2C".

lurch, 2021-04-09:
Yes, this was a side-effect of the way the documentation in this chapter was copy'n'pasted from the Synopsys documentation.
Will be fixed in the next version of the PDFs, thanks.

closed 2021-04-09 by lurch

aallan, 2021-04-09:
"Synopsys Proprietary. Used with permission."

[image]

fivdi, 2021-04-09:
C'est la vie