raspberrypi/pico-feedback #139: SWD max speed ?

Wiz-IO, 2021-04-12:
Hi there,

https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=308588&p=1850544#p1850544

What is max frequency of SWD ( SWCLK & SWD pins )
as experiment work @ 10MHz...

liamfraser, 2021-04-13:
We would recommend a maximum frequency of around 24MHz. The actual max speed you can use reliably depends on the quality of your cables, if you are using a breadboard, etc. OpenOCD tends to default to a low speed of 1-5MHz so that SWD is reliable in most circumstances. You can always experiment and keep increasing the frequency until it is unreliable.

lurch, 2021-04-13:
Should this info be added to our documentation somewhere? Or will the speed of SWD be irrelevant for most users?

Wiz-IO, 2021-04-13:
I test speed with CMSIS-DAM "simulator"
He work on core_1 and simulates the pulses directly on the pins...

no idea ... it is good to have information

aallan, 2021-04-13:
> Should this info be added to our documentation somewhere? Or will the speed of SWD be irrelevant for most users?

Maybe it should go into the Pico Datasheet? Maybe?

UweBonnes, 2021-04-21:
+1 for documenting it in the datasheet.

liamfraser, 2021-11-03:
This will be in an upcoming docs release. Section 2.3.4 Processor Subsystem -> Debug

closed 2021-11-03 by liamfraser

MikeDB1, 2021-11-03:
Can we have the numbers or a pre-release copy of Section 2.3.4 now please ?   We have a design stuck at the moment because of this.

liamfraser, 2021-11-03:
"NOTE: We recommend a max SWD frequency of 24MHz. This depends heavily on your setup. You may need to run much slower (1MHz) depending on the quality and length of your cables." @MikeDB1 if you are using PCB traces then 24MHz is the recommended max. If you are using cables it may be lower.

MikeDB1, 2021-11-03:
Okay that's fine.  The RP2040 is right underneath the CM4 and we can see it work at both 12.288 and 24.576 MHz (professional audio master clock frequencies) but we don't like running out of spec as that is a way to get bitten later.

Many thanks