raspberrypi/pico-feedback #141: Raspberry pi pico datasheet errors

ingand, 2021-04-14:
Hi! in the document "Raspberry pi pico datasheet" there are some errors:
pag 20, in the scheme fig. 16, pmosfet DMG2305UX must be overturned (drain reverse with source); "....P-MOSFET for most situations is Diodes DMG2305UX which has a maximum Vt of 0.9V and Ron of 100 milliohms (at 2.5V Vgs)"
must be changed in "P-MOSFET for most situations is DMG2305UX which has a maximum Vt of 0.9V and Ron of 100 milliohms (at -2.5V Vgs)."
Pag 21, fig. 17 same error su pmos, it must be overturned (drain reverse with source);

lurch, 2021-04-14:
ping @jimbojr and @mike-stimson

daveythacher, 2021-04-27:
Vgs should be negative for PMOS. PMOS is enhancement and oriented incorrectly for PMOS. Symbol is correct for PMOS.

When Vgs is set to -2.5V, the test measures 52m Ohm typical with 100m Ohm max. Id is -3.4A. Vds is expected to be less than -0.25V. Which is mentioned, and is the whole point of this line. I think Vt is mostly not important here.

See datasheet: https://www.diodes.com/assets/Datasheets/DMG2305UX.pdf

May work with body diode to create diode OR, but I would not recommend that. I do not believe this is recommending it either. Example before is better which used discrete diode for diode OR. PMOS may have better drop than schottky which it does mention.

daveythacher, 2021-04-27:
Multiple figures are incorrect.

aallan, 2021-04-27:
> Multiple figures are incorrect.

Which ones?

daveythacher, 2021-04-27:
> > Multiple figures are incorrect.
> 
> Which ones?

Page 20 and 21.

jimbojr, 2021-04-28:
None of the figures are incorrect - the FET is the correct way around for this application. If it was the other way around the body diode would conduct if VSYS was (a body diode drop) higher than the soruce voltage.

Vth matters if you want to make sure your FET turns on sufficiently over the voltage range you expect to see from the source.

No problem to add a -ve sign to Vth if it helps people, though (as it's known to be PMOS) the sign is often omitted.

closed 2021-04-28 by aallan

daveythacher, 2021-04-28:
> None of the figures are incorrect - the FET is the correct way around for this application. If it was the other way around the body diode would conduct if VSYS was (a body diode drop) higher than [...]

Body diode is weak and has high voltage drop more than likely. It would only conduct if the voltage difference from the VUSB and voltage source is significant. In this case then yes a decent current flow may exist which can be bad. This difference must exceed the Vf of body diode.

You are not really interested in the transistor specs because you plan to use the body diode. The body diode runs in reverse of transistor. It can be used for fly back in some cases because of this. The body diode if strong enough can be used as diode however this is not recommended. Also you do not mention how to select this properly.

Transistor will not likely turn on. (Therefore no conduction from switch.) Vgs is likely zero. Min threshold (Vt) for Vgs is -0.5V. Vgs of -2.5V exceeds Vt. The diode begins to conduct at Vt but is very small current flow 250uA. MOSFET is sometimes called voltage controlled resistor. This affects Id and Vds.

Vt in this case is the voltage drop of the diode between VUSB and VSYS. By design this value is likely low, perhaps 0.3 to 0.7V which means the new source would be able to provide maybe 1mA. If in proper configuration, it would default to higher voltage supply. Vgs must pass Vt to cause diode to be on, however to allow significant flow use larger value on Vgs. See datasheet.

Negative is standard for PMOS. Positive is standard for NMOS. PMOS is high side so you are nearly guaranteed it. See inverter. There are a few other things which I do not see here, so I will not mentioned them.

If the assumption is that Vt is Vf then it may still conform to the same forward current. At which point Vgs is the expected Vf? Never heard of this, but just tried to understand how that works. Diodes IV relationship are not fixed or linear. This will create poor efficiency, which was the entire point.

I would recommend only mentioning diode if you plan to use body diode. Body diode is advanced. Further explanation is required.

daveythacher, 2021-04-29:
Actually the difference would need to be body diode drop plus series diode drop. The difference between these power rails is not expected to be this extreme. Body diode from datasheet looks fairly close to standard diode in terms of Vdrop. (Fig 8?)

Worst case is USB with 1.8 power source.

Again this is different from what it is explaining. Both models probably will work here. However this may require closer inspection.