raspberrypi/pico-feedback #142: Invalid decorator documentation in Raspberry Pi Pico Python SDK PDF

lurch, 2021-04-14:
As spotted by https://github.com/micropython/micropython/issues/7117 the documentation on page 19 of  https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf says that the `@asm_pio` decorator takes `set_pins` and `sideset_pins` parameters. But AFAICT these are incorrect and should actually be `set_init` and `sideset_init` ?
I'll update these as necessary after doing the necessary investigation & testing.

kevindawson, 2021-04-14:
@lurch 

a gentle nudge ;)

- ..._init in @asm_pio decorator
- ..._base in StateMachine instantiation

had you considered adding an additional code example inspired by [#6933](https://github.com/micropython/micropython/issues/6933) because it's cool and informative.

lurch, 2021-04-14:
We don't have time to create additional code examples ourselves (too busy with other stuff!) but people are welcome to submit examples to https://github.com/raspberrypi/pico-micropython-examples/

robert-hh, 2021-04-14:
More of my exercises are here: https://github.com/robert-hh/RP2040-Examples

kevindawson, 2021-04-14:
@robert-hh 

I'm now a member of the happy bunny club :)
new more interesting examples - no more re-worked noddy:(
Thanks

aallan, 2021-04-15:
> We don't have time to create additional code examples ourselves (too busy with other stuff!) but people are welcome to submit examples to https://github.com/raspberrypi/pico-micropython-examples/

Yup. We're happy to take PRs for micropython example code to be included into https://github.com/raspberrypi/pico-micropython-examples. Larger examples with an accompanying README.adoc file, e.g. https://github.com/raspberrypi/pico-micropython-examples/tree/master/pio/neopixel_ring, will also get pulled into the PDF documentation.

kevindawson, 2021-04-16:
@lurch as requested:
### Part 1,
from: Raspberry Pi Pico Python SDK
The idea is that for the 4 sets of pins ( in , out , set , sideset , excluding jmp ) that can be connected to a state machine,
there’s the following that need configuring for each set:

μpython-rp2
@asm-pio - contains (set_init, sideset_init, out_init) But No (in_init)
StateMachine - contains ( set_base, sideset_base, out_base, in_base)

As we can use PIO.OUT_LOW, PIO.OUT_HIGH, PIO.IN_LOW, PIO.IN_HIGH

Either:

A) in_init is missing
or
B) out_init, out_base, in_base are superfluous -> μpython bloat 

### Part 2,
after more reading:

- RP2024 Datasheat
  - 3.5.6 GPIO Mapping
    - Figure 48
  - 3.7. List of Registers
    - IN_SHIFTDIR
      - 1 = shift input shift register to right (data enters from left).
      - 0 = to left.
    - AUTOPUSH
      - Push automatically when the input shift register is filled, i.e. on an IN instruction which causes the input shift counter to reach or exceed PUSH_THRESH.
   - 3.5.4.1. Autopush Details

there is pre-state mapping for an input (PIN) set against in_base (ONE pin only Vasili) But you keep using **pins**

### Part 3,
From [pico-micropython-examples](https://github.com/raspberrypi/pico-micropython-examples/blob/master/pio/pio_pinchange.py#L1-L46)
```python
# Instantiate StateMachine(0) with wait_pin_low program on Pin(16).
pin16 = Pin(16, Pin.IN, Pin.PULL_UP)
sm0 = rp2.StateMachine(0, wait_pin_low, in_base=pin16)
sm0.irq(handler)
```

### Conclusion 
for continuity we should be doing this:
```python
@asm_pio(in_init=PIO.IN_HIGH)
```
&#8756;
A) in_init is missing

lurch, 2021-04-16:
> for continuity we should be doing this: `@asm_pio(in_init=PIO.IN_HIGH)`
∴ A) in_init is missing

ping @dpgeorge - should there be an `in_init` in the decorator?

lurch, 2021-06-04:
Reopening, as I've not yet made the corrections mentioned in the first comment in this issue. Another thing still in backlog!

aallan, 2024-03-11:
Fixed in the documentation. Will be present in the next release.

closed 2024-03-11 by aallan