raspberrypi/pico-feedback #143: DWT (Data Watchpoint and Trace) unit?

WestfW, 2021-04-15:
The rp2040 datasheet mentions watchpoints, and "refer to the ARM v6m ARM", which presumably means that it includes the cm0+ DWT unit.  But the SDK doesn't seem to include any register definitions for the DWT at all.

https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=309540&p=1851849#p1851357

(heh.  This could be solved by using the standard core_m0plus.h cmsis header ala https://github.com/raspberrypi/pico-sdk/issues/21 ) (assuming that the rp2040 supports the full DWT.)

kilograham, 2021-04-15:
DWT are included, but are not accessible from executing code, so not included in headers; as per referenced CMSIS headers:

```
/**
  \ingroup  CMSIS_core_register
  \defgroup CMSIS_CoreDebug       Core Debug Registers (CoreDebug)
  \brief    Cortex-M0+ Core Debug Registers (DCB registers, SHCSR, and DFSR) are only accessible over DAP and not via processor.
            Therefore they are not covered by the Cortex-M0+ header file.
  @{
 */
/*@} end of group CMSIS_CoreDebug */
```

WestfW, 2021-04-16:
Fair enough.  ARM ARM says only:
> The DWT, BPU, ROM table, DCB, and debug registers in the SCS are accessible through the DAP interface. Access from the processor is IMPLEMENTATION DEFINED.
I don't really see how any of these are likely to be useful if accessed by the CPU; it doesn't look like CM0+ really supports semi-hosted(?) debugging...

lurch, 2021-04-16:
Can this be closed @WestfW ?  It sounds like the "IMPLEMENTATION DEFINED" you're referring to is part of the Cortex M0+ implementation-details, rather than part of the RP2040 implementation-details?

kilograham, 2021-04-16:
"Implementation Defined" would usually mean up to the RP2040, and the ARM docs are not super clear, but it is actually defined as always no in M0+ (see section 7.2 in the Crotex M0+ Technical Reference manual)

[image]

closed 2021-05-25 by aallan