raspberrypi/pico-feedback #144: Today download install Errors

K2AAE, 2021-04-15:
New PI 4+8GB /128GB USB - Clean with LAMP installed. 4/10/2021 Python 3.9.4
From https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf followed the steps to the point of trying to enter hello_usb but there is a menu difference, I had no drop down list. I Clicked around and ended up building all, but:

-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
ELF2UF2 will need to be built
PIOASM will need to be built
I checked the 'pico/pico_examples/build/hello_world/usb/hello_usb.uf2 was there with 65.5KiB

lurch, 2021-04-15:
Re-opening, because I still want to double-check the VSCode instructions (and screenshots) at some point.

aallan, 2021-05-25:
@lurch Did you take a look, or still in backlog?

lurch, 2021-05-25:
Still in backlog.

SubaruArai, 2022-04-08:
@danielo515 Manually adding include path other than ``CMakeLists.txt`` isn't a good idea, just let [cmake-tools](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cmake-tools) do the config.

Sometimes, vscode will be dumb enough to ignore that info. In that case, [export compile_commands.json](https://cmake.org/cmake/help/latest/variable/CMAKE_EXPORT_COMPILE_COMMANDS.html) and tell the C/C++ extension its path.

example c_cpp_properties.json:
```json
{
    "configurations": [
        {
            "configurationProvider": "ms-vscode.cmake-tools",
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
}
```

danielo515, 2022-04-08:
Ok, thanks.
I will try that next time I can get into a pico project.
Cheers!

lurch, 2025-07-09:
Closing this, as this has all been made **much** simpler (and more reliable) with the release of https://github.com/raspberrypi/pico-vscode (which is now what https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf documents).