raspberrypi/pico-feedback #145: [windows] pico not showing up on any com port

n-mam, 2021-04-20:
I received my pico today and the blink test worked fine.. However I don't see any COMx device being created. This is how my devmgmt looks. This is on Windows 10 (installed around a month back). I couldn't find any RPI KB which highlights which is the exact USB to serial driver needed for this to work.. 

[image]

aallan, 2021-04-20:
> I received my pico today and the blink test worked fine.. 

Not sure what you mean by "blink test", did you install the `blink.uf2` file on your Pico?

> However I don't see any COMx device being created. 

Not sure why you'd expect one.

For instructions on how to get started with Raspberry Pi Pico on Windows 10 with the C SDL, see [_"Getting started with Raspberry Pi Pico"_](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf), especially Section 9.2.

For instructions on how to get started with Raspberry Pi Pico with MicroPython, see [_"Raspberry Pi Pico Python SDK"_](https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf) which walks you through installing Thonny.

n-mam, 2021-04-20:
> Not sure what you mean by "blink test", did you install the blink.uf2 file on your Pico?

yes

> Not sure why you'd expect one.

page 14 of the same guide which you pointed me at we have :

`With your Raspberry Pi Pico connected directly to your Raspberry Pi via USB, see Figure 2, you can see the text by
installing minicom:`  what is that all about then ? Shouldn't one expect an equivalent COM port USB to serial connectivity on Windows ?

lurch, 2021-04-20:
Yes, after flashing the `hello_world.uf2`, but your original post implied you were expecting a COM port from the `blink.uf2`?  Can you give more details about exactly what you've done please? (it's always difficult to try and second-guess)

aallan, 2021-04-20:
> page 14 of the same guide which you pointed me at we have :
>
> `With your Raspberry Pi Pico connected directly to your Raspberry Pi via USB, see Figure 2, you can see the text by installing minicom:` [...]

The `blink.uf2` binary just blinks the LED on and off again. There is no UART, that section (Section 4.3 and 4.4) is talking about the `hello_world.uf2` binary. Scroll down on [https://www.raspberrypi.org/documentation/rp2040/getting-started/#getting-started-with-c](this page) to the "_Say 'Hello World'_" section and download the UF2 file there.

n-mam, 2021-04-20:
we can leave blink aside for a while. I just highlighted that initially to point out that the board was up and running fine on arrival.. nothing more. My question is what triggers the COM port creation like what has been highlighted on page 12 (i.e. besides directly hooking up to the UART pins directly via a USB to serial convertor) ? is it program specific ?

aallan, 2021-04-20:
See Chapter 4 of [_"Getting started with Raspberry Pi Pico"_](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf) for discussion of how to write a program with UART or USB serial enabled.

n-mam, 2021-04-20:
Indeed it does show up on flashing the hello world sample: Thanks for the pointer

[image]

closed 2021-04-20 by n-mam

AdityaC4, 2024-03-11:
For me, it still does not show up even if I use Hello world example. I changed some settings configuring my other devices using Zadig for Pulseview and while doing that my pico was connected and I think I messed up the drivers. Is there a way to fix this?