raspberrypi/pico-feedback #146: Broken serial output

alexandruradovici, 2021-04-20:
I am using a Pico connected with three cables to serial0 of a Raspberry Pi. I tried using several types of Raspberry Pis (2, 3+, and 4), several cables and several Pico Pis. The output from the serial is mostly broken, due to several missing characters. This happens randomly, I am not able to find a rule.

Sometimes resetting the Pi Pico helps.

I tried running the hello_world over serial example.

I was wondering if anybody else had a similar issue?

aallan, 2021-04-20:
> No, I did not know it existed, I used raspi-config to setup the serial port.

When you enabled the serial port, you did ensure to disable the login shell?

alexandruradovici, 2021-04-20:
> > No, I did not know it existed, I used raspi-config to setup the serial port.
> 
> When you enabled the serial port, you did ensure to disable the login shell?

Yes.

alexandruradovici, 2021-04-20:
> If Bluetooth is enabled it could help to explain the problem on a Pi 3 or Pi 4 but not on a Pi 2.
> 
> When using raspi-config to configure serial, it's also important to answer the following question with "No":
> 
> ```
> Would you like a login shell to be accessible over serial?
> ```
> 
> If this question is answered with "Yes", there will be a login shell associated with the UART and it will read some of the serial input that is coming from the Pico.

I answered here with No an rebooted the Raspberry Pi Linux.

fivdi, 2021-04-20:
And `/boot/config.txt` contains the following:
```
dtoverlay=disable-bt
enable_uart=1
```

lurch, 2021-04-21:
@alexandruradovici You mentioned you have multiple Pis - are you able to use a simple 3-wire crossover cable between the UARTs on two separate Pis, and run minicom on each to communicate between them? (which will help narrow down whether the problem is with your Pi or with your Pico)

iuliutza, 2021-04-23:
@fivdi Hi, after adding what you suggested in config.txt it all worked well. Thank you!

lurch, 2021-04-24:
@aallan I'm not sure if this is ready to be closed as @alexandruradovici reported the problem originally, and @iuliutza said that the suggestion worked for him.

fivdi, 2021-04-27:
It looks like all is ok now. See https://github.com/WyliodrinEmbeddedIoT/tock/pull/6.

lurch, 2021-04-27:
@alexandruradovici Can you confirm if this problem is fixed for you, and close this issue if appropriate please?

alexandruradovici, 2021-04-28:
I am sorry for the small delay, it seems to work now. Just to sum of, if others have the same issue:
- *Raspberry Pi* 2 - it requires a reboot from time to time
- *Raspberry Pi3/4* - disabling BLE seems to fix the serial (suggested by @fivdi)

Thank you for the feedback.

closed 2021-04-28 by alexandruradovici