raspberrypi/pico-feedback #148: Cannot open `stdlib.h` on WSL 2 with VSCode/CMake

icorbrey, 2021-04-23:
I've set up my project using [this tutorial](https://paulbupejr.com/raspberry-pi-pico-windows-development/), and everything's gone pretty smoothly. However, once everything was set up I opened up `blink.c` and got the following error when trying to include `pico/stdlib.h`:

```
cannot open source file "pico/version.h" (dependency of "pico/stdlib.h")
```

I haven't been able to find any advice on this issue other than [this post](https://www.raspberrypi.org/forums/viewtopic.php?t=299887) on the Raspberry Pi forums saying I need to change CMake's build generator. Doing so had no visible effect. Any advice?

aallan, 2021-04-23:
> @aallan This should probably be moved to pico-feedback?

Done.

aallan, 2021-04-23:
> I've set up my project using [this tutorial](https://paulbupejr.com/raspberry-pi-pico-windows-development/), and everything's gone pretty smoothly…

That tutorial covers setting up the toolchain on Windows using WSL. We haven't tested that in house, although I *think* one of the engineering team is using the toolchain via WSL. Reading through the post it looks about right, but we don't have a set of official instructions at this point.

icorbrey, 2021-04-23:
Here's my build log:

```
[variant] Loaded new set of variants
[kit] Successfully loaded 2 kits from /home/icorbrey/.local/share/CMakeTools/cmake-tools-kits.json
[main] Configuring folder: pico-examples 
[proc] Executing command: /usr/local/bin/cmake -H/home/icorbrey/pico/pico-examples -B/home/icorbrey/pico/pico-examples/build -G "Unix Makefiles"
[proc] Executing command: /usr/local/bin/cmake --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_C_COMPILER:FILEPATH=/usr/bin/arm-none-eabi-gcc-6.3.1 -H/home/icorbrey/pico/pico-examples -B/home/icorbrey/pico/pico-examples/build -G "Unix Makefiles"
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Configuring incomplete, errors occurred!
[cmake] CMake Error at pico_sdk_import.cmake:44 (message):
[cmake]   SDK location was not specified.  Please set PICO_SDK_PATH or set
[cmake]   PICO_SDK_FETCH_FROM_GIT to on to fetch from git.
[cmake] Call Stack (most recent call first):
[cmake]   CMakeLists.txt:4 (include)
[cmake] 
[cmake] 
```

aallan, 2021-04-24:
> [cmake]   SDK location was not specified.  Please set PICO_SDK_PATH or set

You don't have `PICO_SDK_PATH` set so CMake can't find the the SDK.

icorbrey, 2021-04-24:
@aallan I get that, however I have already set this in 3 different places (VSCode's CMake settings, `.bashrc`, and `.zshrc`). Is there a specific place where I should set it instead?

lurch, 2021-04-24:
@icorbrey Neither myself nor @aallan use WSL, so you'll have to ask the author of the 3rd-party tutorial you're following.

icorbrey, 2021-04-25:
Got it working! Instead of setting the SDK path in _CMake: Build Environment_, I had to set it in _CMake: Configure Environment_. Thank you guys for your help!

closed 2021-04-25 by icorbrey

aallan, 2021-04-25:
> Got it working! Instead of setting the SDK path in CMake: Build Environment, I had to set it in CMake: Configure Environment. Thank you guys for your help!

Yup. See Section 9.2.4 in [_"Getting started with Raspberry Pi Pico"_](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf).