raspberrypi/pico-feedback #149: Question: C++ wrapper in progress?

Reneg973, 2021-04-25:
Hi there,

is there anybody out there who thinks about (or already began) writing a C++ wrapper around this C functions?

lurch, 2021-04-26:
FYI the existing SDK _can_ be used with C++ code - see e.g. https://github.com/raspberrypi/pico-sdk/tree/master/tools/elf2uf2

Although as the existing SDK supports both C and C++, it might not as be as "C++ friendly" as a wrapper which _only_ supported C++ and not C ? :man_shrugging:

Reneg973, 2021-04-26:
> FYI the existing SDK _can_ be used with C++ code - see e.g. https://github.com/raspberrypi/pico-sdk/tree/master/tools/elf2uf2.

Thanks for this extraordinary recentness.

Reneg973, 2021-04-26:
Yes I know all this. I asked for a developer, who already began implementing such a wrapper or maybe one, who is interested in developing one together

fivdi, 2021-04-26:
@Reneg973 Maybe someone over at the pico-sdk forum has more information: https://www.raspberrypi.org/forums/viewforum.php?f=145

Reneg973, 2021-04-26:
@fivdi may be a good idea, thanks