raspberrypi/pico-feedback #150: C SDK documentation: PIO JMP X-- clarification

snoring-pigeon, 2021-04-28:
Suggestion for clarification in datsheet (p.341) and C SDK (p. 59) docs:

JMP X11 is described as "scratch X non-zero, post-decrement"

I find the use of 'post' confusing - it could mean
 1) test if X is non-zero after/post decrementing it (reading as plain English)
 2) test if X is non-zero, decrementing it afterwards/post (reading as C programmer)

I think this could be clearer, e.g.

"scratch X non-xero, decrement after test if X > 0"

(assuming the decrement only happens if X > 0, rather than wrapping to 0xffffffff)

dshadoff, 2021-05-23:
Please update the document; I found both of these phrases to be apparently conflicting/ambiguous as well.
If the "scratch X non-zero, post-decrement" phrase were to be written "scratch X non-zero, prior to decrement" it would lose this apparent contradiction and be much clearer.

lurch, 2022-06-09:
See the suggested re-wording here: https://github.com/raspberrypi/pico-feedback/issues/245#issuecomment-1151484429

lurch, 2022-06-16:
Thanks, will be fixed in the next PDF release of the databooks.

closed 2022-06-16 by lurch