raspberrypi/pico-feedback #151: Typo in mutex documentation

smithps, 2021-04-30:
I wish to report an error in the Pi Pico manual. 

Manual: Raspberry Pi Pico C/C++ SDK Release 1.4.1      

The same Typo error is on Page 200 and 201      

The phrase:    
...then true will be returned and the calling core will own the mutex, otherwise false will be returned and the calling core will own the mutex.      

Should read:    
...then true will be returned and the calling core will own the mutex,   otherwise false will be returned and the calling core will *not* own the   mutex.

Looks like a cut and paste of the first typo.

lurch, 2021-04-30:
This text comes from the doxygen-comments in the SDK, and it's correct at https://raspberrypi.github.io/pico-sdk-doxygen/group__mutex.html#ga69e2f75f0a38378b20c0f2bf9cc7bbe3 and also correct when I _locally_ build `raspberry-pi-pico-c-sdk.pdf`, but incorrect at https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf :confused: 
Very strange, currently investigating...

lurch, 2021-05-04:
The "works for me when I locally build" was actually a red-herring caused by accidental doxygen-version mismatches (I **can** now reproduce this myself). I've tracked down the problem to a bug in our XSLT-processing pipeline...

aallan, 2021-05-25:
@lurch Did this get resolved, or is it in backlog?

lurch, 2021-05-25:
In backlog, is a problem with my XSLT.

lurch, 2022-06-16:
The bug here was that the "not" in "the calling core will not own the mutex" was inside asterisks to make it **bold**, but a bug in my XSLT meant that bold-text wasn't getting included in the PDF output. Whoops!

I can't remember if I fixed that particular XSLT bug (probably not?) but in the meantime @kilograham changed the doxygen-comment to take out the asterisks, so https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf _already_ correctly says "Otherwise (if the mutex was already owned) this will return false and the caller will NOT own the mutex."

So the original issue has been solved, but I'll leave this open to remind myself to fix bold-text in Doxygen output in case this ever occurs again.

kilograham, 2022-06-18:
Yeah, maybe we can bump this in priority :-)

lurch, 2022-12-02:
Closing as invalid, as the SDK API docs are now only at https://raspberrypi.github.io/pico-sdk-doxygen/ and no longer included in the C-SDK PDF.

closed 2022-12-02 by lurch