raspberrypi/pico-feedback #157: Valid second stage flash; checksum

tweiss-tgm, 2021-05-18:
Hello, I hope this question is appropriate here.

I tried to work with the rp2040 without the sdk, just to learn something new, and especially better myself in assembly. I read through the datasheet, but as it seems, I am overlooking something, probably obvious. So as the datasheet stated, while "booting" the bootrom eventually tries to load 256bytes from flash to sram, the second stage flash.

Unfortunately none of the second stage flashes I've written (mostly trying to turn a GPIO high) seem to be valid, as in missing their checksum. As the datasheet says, if no valid checksum is present after the 252bytes of code in the flash, the pico drops in BOOTSEL mode. I am kinda lost with generating a checksum. I've tried to dump the specific section from my elf (with objdump) and simply compute the checksum of these raw bytes (the checksum calculation uses the datasheet-specified params), dump them in their own section in the source code and place them at 0x1000 00FC (aka 252bytes after the start of the flash), regenerate my elf and uf2 and I should be ready to go.

But as it seems, somethings wrong with what I am doing. I "tracked" down this problem, as my first try at getting control of the GPIOs didn't work, as in, of course, I was always dropped in BOOTSEL, but furthermore, injection with gdb ("load") resulted in "me" being unable to change the content of RW registers (SIO, IO_BANK0, etc..).

How would I go about generating the checksum correctly and furthermore not manually?

(I know that the second stage flash should just initialize XIP and or copy content from flash to SRAM, but for now I would just like to be able to produce runnable code, without being connected via the debug pins)

I appreciate any kind of help, ideas, directions! Thanks in advance.

lurch, 2021-05-18:
Does https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/boot_stage2/pad_checksum help?

kilograham, 2021-05-18:
Also it is OK to bolt a compiled boot_stage2 from the SDK onto the front of your binary too :-)

tweiss-tgm, 2021-05-19:
Hello!

Thank you very much, I highly appreciate the answers. 

First I tried the pad_checksum script, which unfortunately didn't seem to bring me closer to having a working stage2. After that, I just wanted to see, if a precompiled stage2, taken from the pico-sdk's blink example, would work, which in fact does work.

I compared the sections of the generated elf and they seem to match (at least address, offset, flags, alignment, etc.). I am quite uncomfortable with seeking help for my amateur tries of getting something to work, but again, if any of you have any clues, etc, I would be really grateful.

What I have done so far:
+ create a small stage2, which should configure SIO and a specific GPIO
+ create a linker script, so the code / sections get loaded at the right addresses
+ generated an elf file, dumped the boot2 section, padded and checksum(ed) it with the "pad_checksum"
+ assembled and linked the new source file (from pad_checksum) again into an elf
+ compared the properties of the section with the provided blink-examples boot2
+ generated an uf2 file, with the elf2uf2 tool from the sdk
+ deployed the uf2, and to my unfortune (?) I get dropped into BOOTSEL.

I can provide source files if any of you wants to see them, but I dont think that they are the problem per se. I should probably just use a precompiled boot2 section, but I feel like cheating myself, especially as I wanted to solve these kind of problems by myself.

Thanks!

tweiss-tgm, 2021-05-20:
So I was too naiv in thinking everything works out of the box as I want it to be, I, of course, didn't check the seed the pad_checksum script was using by default (0x00). The Pico needs a seed of 0xffffffff, so pad_checksum needs this as the seed, to create valid stage 2 flashes.

I am just writing this, if anybody finds themselves in similar troubles. Thanks again.

closed 2021-05-20 by tweiss-tgm

lurch, 2021-05-20:
> The Pico needs a seed of 0xffffffff, so pad_checksum needs this as the seed, to create valid stage 2 flashes.

Ah, [so it does](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/boot_stage2/CMakeLists.txt#L69)! I hadn't noticed that :slightly_smiling_face: