raspberrypi/pico-feedback #161: Documentation in formats other than PDF

czak, 2021-05-24:
Would it be possible to release the Pico/RP2040 documentation in other file formats?
There's some great content in the datasheets and the SDK/hardware docs, it's just a bit hard to read on a smaller screen.
I'd love a format which can be reflowed/resized more easily.

There was some discussion about ePub earlier in https://github.com/raspberrypi/pico-feedback/issues/13, but I don't think they were produced, or maybe I missed that?

aallan, 2021-05-24:
So yes, there are ePub versions of the RP2040 and Pico PDF documentation. But, they (mostly) don't pass ePub Lint, so we've been reluctant to publish them as a result. They're also fairly bulky, and therefore doesn't play well on older devices. 

Given those constraints, they do work okay on my iPad and other devices that handle ePub natively, but ePub -> mobi fails fairly spectacularly so they're not currently available on Kindle. 

This is all sortable, at least in theory, but there didn't really seem to be a demand.

czak, 2021-05-25:
My primary use case is an iPad mini.
I would gladly take an ePub, production-ready or not.
But I will admit I'd probably just unpack it and consume as HTML from a home server.

I do understand you wouldn't want to publish something up to par though.
Do let me know if I can help move it forward.

aallan, 2021-05-25:
> My primary use case is an iPad mini.

I've tested on an iPad mini 4. They load, but page display for some of the documents is very slow. I think it's sortable, mostly through optimisation — we have quite a lot of figures where PNGs or JPGs are used rather than a vectorised format such as SVG due to the nature of the materials, e.g. screen shots — to reduce the size of the files used in the figures (but only for the ePubs where it really matters due to reflow).

> But I will admit I'd probably just unpack it and consume as HTML from a home server.

There is the possibility that there will be HTML native versions of the documentation, potentially Q3 or Q4 this year after another related documentation project completes. But no promises.

> I would gladly take an ePub, production-ready or not.

Apparently Github doesn't let you attach an ePub file to an issue (who knew?), so I'e uploaded it and the current documentation release (V1.4.1) as ePub can be found over [on Google Drive](https://drive.google.com/drive/folders/1k7egaE2fp3U7EGsnkqmDCiyc6_4DEJVP?usp=sharing). 

> Do let me know if I can help move it forward.

I'd be interested to know how you get on with it, so please report back on installation issues and how useable the ePubs are in their current state to help inform the decision about whether we should spend the time to get them into better shape.

czak, 2021-05-25:
Thank you for sharing the files.

I've tested the ePubs and now I understand what you meant by "bulky".
My iPad Mini is gen. 2 (2013) and it struggled.

With `rp2040-datasheet.epub`:
* In Apple Books, the doc renders nicely, but flipping pages is really slow. To the point of making the app unresponsive.
* In TotalReader, the doc loads and seem to render fine, but crashes the app after a few pages.
* In FBReader, the doc loads quite quickly and would be usable, but the app has very limited HTML formatting (no SVG, no HTML), so the document is not what it should be (probably similar to what you get with a mobi conversion)

tl;dr The ePubs in their current state are not usable for me.

> There is the possibility that there will be HTML native versions of the documentation

If I had a vote, I'd vote for this instead of the ePubs. I actually think the HTML inside these ePubs is in a good state already. The markup looks decently structured and with just a minor CSS tweak I got it to **very readable** for my device:

[image]

The iPad still struggles a bit when rendering the pages (I think it runs out of memory), but I wouldn't mind reading it like that for a while.

aallan, 2021-05-25:
> I've tested the ePubs and now I understand what you meant by "bulky".

🤣 

I did warn you!

> tl;dr The ePubs in their current state are not usable for me.

Thanks for the feedback, this wasn't entirely unanticipated. As I said, i think there is (a lot) room for optimisation, but we haven't looked at it very hard quite yet because there didn't seem to be a demand.

aallan, 2021-09-08:
Moved to the internal ticket system. It's in the queue!

closed 2021-09-08 by aallan

dltech, 2023-04-10:
Could you share with sources of rp2040 pdf datasheet in order to translate it to russian? Some years ago I actually translated speex doc, and do not want to stop.

aallan, 2023-04-10:
> Could you share with sources of rp2040 pdf datasheet in order to translate it to russian? Some years ago I actually translated speex doc, and do not want to stop.

No, sorry. We will not be releasing the source.

Additionally, the RP2040 Datasheet has been released under a CC BY-ND license, translation is viewed as a derivative under copyright law, and this license therefore does not give you the right to publish a translation of the work.