raspberrypi/pico-feedback #162: SPINLOCK registers are marked as RO in SVD and Reference manual, should be RW

9names, 2021-05-27:
In RP2040 Datasheet in section 2.3.1.7 table 79 (page 60), SPINLOCK0-31 registers are listed as being RO (read only).
This is also true inside the SVD.
In the description cell there is the following text:
```
Reading from a spinlock address will:
- Return 0 if lock is already locked
- Otherwise return nonzero, and simultaneously claim the lock
Writing (any value) releases the lock.
```
Unless I'm misunderstanding, that last line means these registers should be listed RW.

lurch, 2021-05-27:
This is similar to #110  (both of these link to our internal issue `SPIV-60`)

aallan, 2021-05-27:
@lurch I'll assign this one to you alongside #110 at that point?