raspberrypi/pico-feedback #163: Advice on using tinyusb is incomplete.

PeterOGB, 2021-05-27:
Section 2.3.6  TinyUSB Port

"The  tinyusb_dev  or  tinyusb_host  libraries  within  the  SDK  allow  you  to  add  TinyUSB  device  or  host  support  to  yourapplication by simply adding a dependency in your executable in CMakeLists.txt"

It should also mention the need to add
target_include_directories(your_target_name PRIVATE ${CMAKE_CURRENT_LIST_DIR})
to your CMakeLists.txt otherwise the tusb_config.h file in the source directory won't be found.  I wasted an hour or so due to this omission !

PeterOGB, 2021-05-27:
> I mean arguably, we didn't tell you to create a `tinyusb.h` file either... maybe we can link to an example

Yes, it's the "this is all you have to do" nature of the text that is misleading.  The example in pico-examples does include a tinyusb.h file so it isn't hard to work out you need one, but the addition to CMakeLists.txt needed to make the #include work  is less obvious and I think should be explicitly mentioned rather than it being suggested that nothing more than adding the dependency is required.

kilograham, 2021-10-21:
tweaked the text

kilograham, 2021-10-21:
fixed in upcoming doc release

closed 2021-10-21 by kilograham