raspberrypi/pico-feedback #164: multicore_fifo_wready function - documentation query

Gerriko, 2021-05-30:
The current Raspberry Pi Pico C SDK documentation states that the **multicore_fifo_wready** function returns true if the write buffer is full. However, upon testing, it looks like it returns true if there is space available inside the FIFO buffer and false if it is full.

Maybe someone can confirm.

fivdi, 2021-05-30:
Your conclusion appears to be correct.

Hers's the code and documentation for multicore_fifo_wready (see [here](https://github.com/raspberrypi/pico-sdk/blob/1.1.2/src/rp2_common/pico_multicore/include/pico/multicore.h#L87-L91)):
```c
/*! \brief Check the FIFO to see if the write FIFO is full
 *  \ingroup multicore_fifo
 *
 *  @return true if the FIFO is full, false otherwise
 */
static inline bool multicore_fifo_wready(void) {
    return !!(sio_hw->fifo_st & SIO_FIFO_ST_RDY_BITS);
}
```

However, the RP2040 Datasheet documents the RDY flag in the FIFO status register as follows:

```
Value is 1 if this core’s TX FIFO is not full (i.e. if FIFO_WR
is ready for more data)
```

lurch, 2021-05-30:
Already fixed by https://github.com/raspberrypi/pico-sdk/pull/239

closed 2021-05-30 by lurch