raspberrypi/pico-feedback #167: Pico BOOTSEL Schematic Incorrect

pidd-wbiz, 2021-06-07:
I believe the Pico schematic in the Pico Datasheet is incorrect for the BOOTSEL switch circuit.

Currently the schematic shows the flash chip (U3) not selected (ncs = high) when the BOOTSEL switch is not pressed - and subsequently the flash chip enabled (nsc = low) when BOOTSEL is pressed.  This must be back to front from reality.

lurch, 2021-06-07:
I'd be _very_ surprised if the schematic is wrong. `nCS` (pin 1) on the Flash chip is also connected to `QSPI_SS` (pin 56) on the RP2040.
... Aha, see section "2.8.1. Processor Controlled Boot Sequence" of the RP2040 datasheet :slightly_smiling_face: 
`Check if SPI CS pin is tied low ("bootrom button"), and skip flash boot if so.`

pidd-wbiz, 2021-06-07:
> I'd be _very_ surprised if the schematic is wrong. `nCS` (pin 1) on the Flash chip is also connected to `QSPI_SS` (pin 56) on the RP2040.
> ... Aha, see section "2.8.1. Processor Controlled Boot Sequence" of the RP2040 datasheet 🙂
> `Check if SPI CS pin is tied low ("bootrom button"), and skip flash boot if so.`

Ah, I didn't realise QSPI SS was a bus label and went off somewhere else (obvious now of course). R11 will allow the switch to be over-ridden by QSPI SS.  It is still a  weird counter-intuitive configuration.

closed 2021-06-07 by pidd-wbiz

lurch, 2021-06-07:
> R11 will allow the switch to be over-ridden by QSPI SS.

That's probably so that (accidental?) presses of the button get ignored if the RP2040 is actively driving QSPI_SS? :shrug: 

You may also find https://github.com/raspberrypi/pico-examples/blob/master/picoboard/button/button.c interesting?

pidd-wbiz, 2021-06-08:
> > R11 will allow the switch to be over-ridden by QSPI SS.
> 
> That's probably so that (accidental?) presses of the button get ignored if the RP2040 is actively driving QSPI_SS? 🤷
> 
> You may also find https://github.com/raspberrypi/pico-examples/blob/master/picoboard/button/button.c interesting?

Yes, thanks, looks like you foresaw my wish to use the BOOTSEL button in my code, my code will be writing to flash (eventually) so I will have to careful.  I was also looking at [https://github.com/jasongaunt/rp2040-bootsel-reboot-example](url) with a view to using that while developing.