raspberrypi/pico-feedback #168: Do we need a specific repo for community-written PIO libraries ?

mjcross, 2021-06-20:
I saw a few references to 'PIO Library' in the docs, but I don't really see one except for the  examples and extras.

Increasingly I should imagine there will be PIO Libraries for specific IO tasks (e.g. I've just written one to [transmit NEC IR codes](https://github.com/mjcross/rpi-pico-nec-ir-transmit-pio)); and it would be nice to have a centralised place to post them.

lurch, 2021-06-20:
PRs to https://github.com/raspberrypi/pico-examples (to the `develop` branch) are more than welcome :slightly_smiling_face: (e.g. https://github.com/raspberrypi/pico-examples/pull/126 )

If you want to include a `README.adoc` like https://github.com/raspberrypi/pico-examples/tree/master/gpio/hello_7segment then we might also include it the "App Notes" appendix of https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf

mjcross, 2021-06-20:
OK I'll have a crack at that 👍 
In order to make it a bit more suitable for pico-examples I'll see if I can add a receive library too using the other PIO instance, so that people can do a sort of IR loopback test.

pmarques-dev, 2021-06-23:
This is my first contribution to pico-examples and didn't notice the develop branch, so I did the pull request against master. Should I just cancel this pull request and post a new one for develop?

lurch, 2021-06-23:
@pmarques-dev There was an SDK release not long ago, so `develop` is currently in-step with `master`. So you should just be able to click the "Edit" button at the top of your PR, and change the base branch (or whatever GitHub calls it) to `develop` :slightly_smiling_face:

pmarques-dev, 2021-06-23:
OK, I've just done that now, thank you

aallan, 2021-06-23:
If there's a PR in place over on `pico-examples` I'll close this issue.

lurch, 2021-06-23:
@aallan There's a PR which @pmarques-dev created, but not yet a PR that @mjcross created (unfortunately there's been a bit of overlap in discussion of the PRs / issues).

aallan, 2021-06-23:
Ah gotcha.

mjcross, 2021-06-23:
> @aallan There's a PR which @pmarques-dev created, but not yet a PR that @mjcross created (unfortunately there's been a bit of overlap in discussion of the PRs / issues).

Thanks @lurch - it's coming :-)
I want to add NEC receive before I create the PR; first iteration of the Rx code in testing this afternoon

mjcross, 2021-06-26:
Update: code is all now working to my satisfaction; I'm just about to fork the develop branch of https://github.com/raspberrypi/pico-examples and start (attempting to) integrate it...

mjcross, 2021-06-26:
Update: code builds within the source tree; now working on a README.adoc

lurch, 2021-06-28:
I'm going to close this issue, now that you've opened https://github.com/raspberrypi/pico-examples/pull/129 :slightly_smiling_face:

closed 2021-06-28 by lurch

mjcross, 2021-06-28:
Excellent! Thanks @lurch that was a job I meant to do this morning but you beat me to it