raspberrypi/pico-feedback #172: Typo in datasheet? (USB)

urish, 2021-07-05:
The table describing the SIE_STATUS (table 414/page 422) calls the LSB "VBUS_DETECTED":

[image]

However, further references to the same bit call it VBUS_DETECT (e.g. bit 11 in INTR/INTE/INTF/INTS registers, pages 431-434):

[image]

Datasheet version:
build-date: 2021-06-23
build-version: ced2189-clean

aallan, 2021-07-05:
Ping @liamfraser

liamfraser, 2021-07-06:
Rather than change either of the register names (affecting software) the simplest thing is to update the reference of the VBUS_DETECT interrupt to say SIE_STATUS.VBUS_DETECTED instead of the incorrect SIE_STATUS.VBUS_DETECT

aallan, 2021-07-06:
> Rather than change either of the register names (affecting software) the simplest thing is to update the reference of the VBUS_DETECT interrupt to say SIE_STATUS.VBUS_DETECTED instead of the incorrect SIE_STATUS.VBUS_DETECT

Is this a TSV thing, or is it a manual Asciidoc fix?

lurch, 2021-07-06:
It's a TSV thing. If @liamfraser is to busy to fix it himself, feel free to assign it to me and I'll add it to my backlog.

liamfraser, 2021-07-06:
I'll sort it now

liamfraser, 2021-07-06:
Closing this issue as it will go into the next datasheet / sdk release

closed 2021-07-06 by liamfraser