raspberrypi/pico-feedback #174: I2C Default GPIO Pins TYPO in Pinout

kevindawson, 2021-07-08:
In **raspberry-pi-pico-python-sdk.pdf**

- Section 3.6 
  - Table 2. Default I2C pins
    - I2C0 SCL Pin 9
    - I2C0 SDA Pin 8

[image]

Your Pinout should follow your SDK

first highlighted this in [#7476](https://github.com/micropython/micropython/issues/7476) 

You need to Correct your Pinout Diagram

aallan, 2021-07-08:
> Your Pinout should follow your SDK

It does, the default pins are shown for the C SDK. 

> You need to Correct your Pinout Diagram

We do not. The pinout diagram is **not** Python-specific. That said, the default Python I2C pins are indicated correctly; with GP8 as a I2C0 SDA(physical pin 11), and GP9 as a I2C0 SCL (physical pin 12). I don't know, or at least I can't remember, why those pins got picked for the Python port.

closed 2021-07-08 by aallan

kevindawson, 2021-07-08:
@JamesH65 any thoughts

JamesH65, 2021-07-08:
I think thinking is so important. But apart from that, as @aallan is in charge of technical documentation at pi, no.