raspberrypi/pico-feedback #175: IRQ C/C++ SDK booklet section typos

matiasilva, 2021-07-13:
Hello o/,

While working on the examples, I noticed a few things in section 4.1.14. I had a brief glance in the other issues and closed ones to see if anyone had addressed this before, but seems not.

* "That all IRQ APIs affect the executing core only" is this sentence meant to just stand on its own?
* "Interrupt NumbersInterrupts" right above the table. This is a greater issue as any subsequent references to the "Interrupt Numbers" table don't go anywhere when clicked
* "Force an interrupt to pending" in section 4.1.14.3.10. I know this is for `irq_set_pending` but the wording feels off..
* "Set specified interrupts priority." should this be possessive?

lurch, 2021-07-13:
> * "That all IRQ APIs affect the executing core only" is this sentence meant to just stand on its own?
> * "Interrupt NumbersInterrupts" right above the table. This is a greater issue as any subsequent references to the "Interrupt Numbers" table don't go anywhere when clicked

These are both my fault - the "upstream source" for that is the Doxygen comments in https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_irq/include/hardware/irq.h
You can see at https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__irq.html#details how that text is _supposed_ to render, however there are a number of deficiencies in the hand-written XLST file that I've created (that converts the Doxygen output into the source for the C SDK databook) that I've not yet had time to fix.
Thanks for spotting these (new to me) problems, I'll add them to the list! (@aallan Another task on my backlog)

> * "Force an interrupt to pending" in section 4.1.14.3.10. I know this is for irq_set_pending but the wording feels off..
> * "Set specified interrupts priority." should this be possessive?

Please feel free to file a PR against https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_irq/include/hardware/irq.h to improve the wording here, and it'll eventually work its way through the system and end up in the PDFs :slightly_smiling_face:

matiasilva, 2021-07-15:
PR submitted, approved and merged. So 1/2 of this issue is solved :D

lurch, 2022-12-02:
Closing as invalid, as the SDK API docs are now only at https://raspberrypi.github.io/pico-sdk-doxygen/ and no longer included in the C-SDK PDF.

closed 2022-12-02 by lurch