raspberrypi/pico-feedback #176: Lack of information for people to use a different build system

GorgonMeducer, 2021-07-14:
Hi ,

With the discussion in [issue506](https://github.com/raspberrypi/pico-sdk/issues/506) it is worth creating a dedicated thread for documentation.

the SDK manual is entirely focusing on CMake and provides limited information for people who want to use a different build system. For example, I thought Appendix D: Board Configuration might help, but it doesn't. It is sad.

If there is any chapter to help people understand the compilation relationship and how the header files are included (obviously there are plenty of decision makings are done by CMake), it will help a lot.

Don't you think people would like to try SDK on IAR, LLVM, arm compiler 5 and arm compiler 6? Not only to mention that the current version of arm GCC is way behind the LLVM, armcc and armclang in terms of performance and code size.

Cheers

aallan, 2021-07-14:
While we're happy for the community to support alternative build chains, and language ports, we don't really have the available personnel to support other toolchains. I'll put this on the list for consideration, but it's unlikely to be something we're going to prioritise when expanding the documentaiton around RP2040.

GorgonMeducer, 2021-07-14:
I can understand that. With the nature of gcc, porting to a toolchain with compatible syntax frontend is relatively easier for open source community, i.e. LLVM or armclang. I am doing this:

https://github.com/GorgonMeducer/Pico_Template

And it works to some degree. 

So, you don't have to add support for different toolchains. If you can add information like:

- compilation relationship and 
- how the header files are included (obviously there are plenty of decision makings are done by CMake)

Cheers,

kilograham, 2021-07-14:
https://github.com/raspberrypi/pico-sdk/issues/196 is tracking CLang support which will probably appear at some point... this is *entirely* orthogonal to CMake.

if you want to use something other than CMake then you are on your own. The CMake build _is_ the documentation. There is this example that gives you a good idea of the include paths (not really that surprising, it is pico/xxx/include for pico_xxx") and required #defines.

https://github.com/raspberrypi/pico-playground/blob/master/standalone/static_sdk/CMakeLists.txt

JamesH65, 2021-07-14:
If you look at <sdk>/cmake/preload/toolchain you will see the mechanism for adding new toolchains. Still a work in progress but it should be enough to get you working with alternative toolchains like clang.

Moving away from CMake would not be so easy, and isn't really necessary. CMake works well, and doing it with another make system would not be easy at all. Lots of professional users are more than happy with cmake, implying that it is not a professional system is wrong.

GorgonMeducer, 2021-07-14:
If my wording gives the impression that I implied using CMake is not a professional system, then that's not what I intended. And I am sorry to give you that impression. 

I understand I am on my own. I am simply not good at cmake and my working environment is dominated by IDE where usually how the project is organised is hidden. And to be honest, please don't assume everyone wants to spend time on understanding how cmake is used. I totally don't care, maybe I am not a good programmer, but for me, a tool should be as simple as possible with reasonable input and reasonable output. And the definition of simple is relevant to whether you have to put extra time to learn new stuff. I want to enjoy life, for me, knowledge of C, compiler and some stupid IDE is good enough to finish daily work. When I fall in love with Pico, I know I have to bring it to my own working environment, and I know it is time-consuming. 

Bringing pico-sdk to a different environment might not be a reasonable behaviour in terms of cost, but being able to use something nice in a familiar environment is something you cannot stop yourself from doing so. 

Maybe there are people like me around the world. 

Thank you for all your suggestions.

GorgonMeducer, 2021-07-14:
When you say: The CMake build is the documentation.  I respect you. So, a person without the knowledge of cmake is not capable to read the book written in cmake.  And this statement simply says, it's there, if you cannot read it, it's your own problem. 

It is my own problem. I accept it. 
You can close this issue as everything is well documented in the cmake script and anyone could find what they want by reading the code.

kilograham, 2021-07-14:
yeah, we chose CMake because (apart from cross platform, trivial IDE/code completion/build integration etc) it makes it easy to compose dependencies _and_ support custom builds of all the libraries for each target (which we want for speed/size/customization)

This flexibilty does not really translate (at least nicely) into raw `make` at all. Therefore if you are using make you are probably also perhaps only building a fairly static set of targets... 

In this case you can use the thing linked above, or `make VERBOSE=1` (with the a prorototype CMake build) to figure out what compilation flags, and link flags the CMake build would use for your particular target. Documenting what the CMake does so that people can not use the CMake is a huge amount of work.

There is definitely some value in adding some more documentation related to the CMake build perhaps (beyond what should be straightfoward with a basic understanding of cmake), and perhaps thats what this issue should refer to

lurch, 2021-07-14:
> a tool should be as simple as possible with reasonable input and reasonable output.

With the amount of flexibility that the Raspberry Pi Pico SDK includes, I think the CMake trade-off is that the "internals" are fairly complicated (which matters to people like you who don't want to use cmake), while having fairly simple "input files" [like this](https://github.com/raspberrypi/pico-examples/blob/master/adc/hello_adc/CMakeLists.txt) (which matters to the vast majority of people who are happy with using our supplied and documented toolchain).
All IMVHO of course.

GorgonMeducer, 2021-07-14:
I understand the trade-off. And If I stand at your position, I would choose cmake also. 
What I have tried is getting help from you and you did provide useful information. I appreciate that. 

If possible, I will raise more questions to help my understanding. If you can help me on that when you have time, that will help me a lot. I believe this is a reasonable way to support people who want to do their own build system. 

Thank you.

In the end, when it is possible, please consider adding more information about the SDK compilation in the future. 
By the way, I just learn that IAR now support debugging of RP2040, so it start another possibility in terms of toolchain for pico-sdk.

lurch, 2021-07-14:
Thank you for your understanding. Makes a nice change from the people demanding that we add support for their favourite IDE / toolchain of choice :slightly_smiling_face:

GorgonMeducer, 2021-07-14:
It is an open-source project and the possibility is unlimited. I know the amount of work is huge. 
Honestly speaking, Raspberry Pi (and Pico) comes from Linux ecosystem, and cmake is just air and water. But Pico is a micro-controller, the bare-metal / RTOS ecosystem have different knowledge base. I know a lot of firmware programmers usually work with MDK, IAR and or makefile system in windows dominated environment.

TrustFirmeware-M had encountered similar problem which makes it very hard in terms of adoption. In the end, TF-M creates CMSIS-Pack and solved the problem. 

Please seriously consider this problem that Micro controllers have a different ecosystem than Linux.

Cheers,
Gabriel

lurch, 2021-07-14:
> In the end, TF-M created CMSIS-Pack and solved the problem.

See https://github.com/raspberrypi/pico-sdk/pulls?q=is%3Apr+cmsis

> Please seriously consider this, that Micro controllers have a different ecosystem than Linux.

Indeed, that's why we spent a fair chunk of time writing instructions for Linux, Windows and Mac in https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf

JamesH65, 2021-07-14:
It's worth noting though, that the Raspberry Pi itself (Pi4, Pi400) are perfectly competent development systems for Pico, and of course they run Linux. It would have been a very odd choice for us to ignore our own platform when developing the Pico toolchain and exclusively target, for example, Windows. As it is we target Linux first, and then wrote up instructions on how to do the same thing in Windows and Mac. 

Out of interest, what do you expect to gain from using a different system to CMake? On the whole to build for Pico requires very little CMake knowledge, because it just works. Put it another way, what isn't CMake doing that you need? What is now difficult for you to do because of the use of CMake that was easy in whatever other build system you use?

GorgonMeducer, 2021-07-14:
I understand your choice. And you don’t need to persuade me to use cmake, because there is less reasonable thing behind but some emotional thing which is, it looks too complicated. 

I have followed the getting started with pico, it is nice. But it doesn’t support Arm Compiler 6. Gcc is less efficient than armclang and IAR. Here is my chain of thought:

1. I want to use Arm Compiler 6
2. Pico-SDK written in GCC with cmake
3. I don’t know cmake well, so I don’t know how to make it work with armclang
4. I know how to use armclang and how to port code written in gcc to armclang
5. I choose MDK because it supports armclang. 
6. I manage to get MDK project work with pico-sdk
7. I just learned that my project configuration didn’t have stage2 boot, so I tried to know how to bring stage2 boot to my project configuration.
8. I have read the SDK manual first, which provides some information but not sufficient
9. I decide to get help from here.

lurch, 2021-07-14:
> By the way, I have ULINKpro and DSTREAM which are supported by MDK (and Arm-DS). I only have one Pico.

You could buy almost 300 Raspberry Pi Picos for the cost of a single ULINKpro. Just saying :wink:

JamesH65, 2021-07-14:
You only need to buy one more to get a decent debugger, by using the second Pico as the debugger...
See appendix A in the following

https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf

and

https://www.digikey.be/en/maker/projects/raspberry-pi-pico-and-rp2040-cc-part-2-debugging-with-vs-code/470abc7efb07432b82c95f6f67f184c0

JamesH65, 2021-07-14:
I wonder is there is a bit of a sunk cost fallacy going on here. You have some expensive debuggers, so you want to use them. But that may not be cost effective. You can get similar levels of capability by buying another $5 Pico and using that.  It will be quick to set up, a half day at the most, unlike these rather expensive debuggers, so its actually cheaper to forget those, and just use Pico's. It does sound counterintuitive when you have already spent a load of money on debuggers, but that's the sunk cost fallacy in a nutshell,.

aallan, 2021-07-14:
> …you don't have to add support for different toolchains. If you can add information like:
> 
> * compilation relationship and
> * how the header files are included (obviously there are plenty of decision makings done by CMake)
> 
> or to make it more general, some insight about the pico-sdk. It helps a lot for professional users.

You've kicked off some internal discussions, and we're going to have a think about things. Especially with the folks at ARM taking a serious look at Clang. Thanks for your input here, it's good to get alternative points of view on how the SDK is being received. That said, to manage expectations, we're a small shop and we're stretched.

GorgonMeducer, 2021-07-14:
Lol. I am glad my input can kick off some discussion. 
ULINKpro and DSTREAM come for free ( they belongs to the Arm office ). So extra $5 is extra. But you are right, it is the best solution, and a new pico is on its way.

Having pico works for MDK brings me extra value because I want to use Pico as a demonstration platform for some proof of concept, for example, [a watch-panel demo](https://github.com/ARM-software/EndpointAI/tree/master/Kernels/Research/Arm-2D/examples/watch_panel)

[image]  

For now, I will keep using MDK (for compilation) and use the drag&drop method for download. 
The key is not that I want to stick to MDK; the key is that I want to use Arm Compiler 6, and I don't have to wait for the pico-SDK team to add the support (it appears that you are lack resources).

Cheers

aallan, 2021-07-14:
> The key is not that I want to stick to MDK; the key is that I want to use Arm Compiler 6, and I don't have to wait for the pico-SDK team to add the support (it appears that you are lack resources).

Yup. Sorry, but we're a small team and lack the resources for that. It's not currently on the roadmap.

JamesH65, 2021-07-14:
It might be worth having a play with the various toolchain options that I mentioned in comment https://github.com/raspberrypi/pico-feedback/issues/176#issuecomment-879856776

Doesn't seem too difficult to add a new compiler toolchain, but I am not sure if the infrastructure is all in place.

GorgonMeducer, 2021-07-14:
Hi aallan,

I have no intention to blame the team. I totally understand you are a small team. That's why I tried to add the armclang support by myself.
So far, I have passed the stage2 boot (verified), the Pico can recognise the image I provided (the mass drive is removed right after the drag&drop).

Unfortunately, the blinking demo doesn't work. I am working on that.

GorgonMeducer, 2021-07-14:
Hi JamesH65,

About the infrastructure, here is a rough list about what I have done:

- Use [c startup code](https://github.com/GorgonMeducer/Pico_Template/blob/main/project/mdk/startup_RP2040.c) to replace the assembly version 
- Add a fake header file for <sys/cdefs.h> 
- Add [implementation for panic](https://github.com/GorgonMeducer/Pico_Template/blob/main/project/mdk/wrapper/env_wrapper.c)
- Redefine macros for armclang
    * offsetof
    * __STRING
    * __force_inline
    * __always_inline
    * __unused
    * __packed_aligned
    * Customise platform.h 
- use "-include" to include a header file that contains
    * stdint.h
    * stdbool.h
    * assert.h
    * static inline function __get_current_exception()
    * definition of linker generated variable: __StackOneBottom
- Copy config_autogen.h to MDK project
- write [scatter script](https://github.com/GorgonMeducer/Pico_Template/blob/main/project/mdk/RP2040.sct)

I hope this helps.

GorgonMeducer, 2021-07-16:
Hi guys,

I want to thank you all for the support you have provided these days. 
An MDK project template is created and it works well.

https://github.com/GorgonMeducer/Pico_Template/releases/tag/v0.9.0

I think I have got what I want.

When I do retrospection, I have to say: there isn't much complexity there that you have to use cmake.
Since I am using a very traditional way to create the project, and MDK is capable to generate a windows batch file, so I believe anyone who wants to use a different build system could use my project as a reference. 

I am done with this issue. Please feel free to close it.

Thank you very much.

Cheers,
Gabriel

kilograham, 2021-07-16:
The problem is that you have just made a brittle simulacrum of the actual cmake build/and copied and pasted bits of code from the SDK. This is going to make people's experiences miserable the moment they try and do anything of your very narrow happy path.

GorgonMeducer, 2021-07-19:
That's true. And when people experienced those miserable, they will come back to cmake build system. I think I have just helped you to prove you are right. This task wasted my own time and life, and people have options to try what they want. 

It's fair for everyone. There is no cheating, right?
I do everything under the BSD-3-clause. If you think there is anything improper, I will correct them the moment I know.

GorgonMeducer, 2021-07-19:
The whole pico-sdk is designed with the help of cmake. In order to compile pico-sdk, I have to mimic the behaviour of the cmake procedure. It is a practice way. This is the truth and I will not put any emotional description about that. 

Another truth is that I provided non-official support for Arm Compiler 6 in my template for pico-sdk. It uses a submodule that respects the original content with no direct modification.

GorgonMeducer, 2021-07-19:
I didn't ruin anyone's work or work experience.

KubaO, 2022-06-19:
Cmake takes several weeks of hard concentrated work to figure out, but once that hurdle is cleared, the system is easy to use, and everyone and their dog integrates with it. Basically, if you can build C using cmake, your platform is automatically supported by every IDE that supports cmake. Cmake feeds the IDE with information about include paths and everything else an IDE needs to enable code insights, real-time diagnostics (on the bare code, before compiling), etc.

I would say that any project that is not using cmake today is making itself harder to become widely adopted.

Interestingly enough, there's some obscure make syntax that translates to obscure cmake syntax directly: `$()` in gnu make translates to `${}` in cmake. What is that, you ask? It's a token separator that doesn't generate a space. Make/cmake see two words on the sides of the separator, but no space is inserted. A rather crucial trick that I've had to use several times, and there was no realistic workaround if it wasn't there.

aallan, 2022-06-19:
As I said further up thread. While we're happy for the community to support alternative build chains, and language ports, we don't really have the available personnel to support other toolchains. It's on the list for consideration, but it is not something we're going to prioritise when expanding the documentation around RP2040.

closed 2022-06-19 by aallan