raspberrypi/pico-feedback #177: A few more SDK typos/inconsistencies

matiasilva, 2021-07-16:
A few more things I've noticed. I will submit a PR for the ones I can fix.

* in section 4.4.3 on `pico_binary_info` there seems to be a) inconsistent spacing for the macros and b) inconsistent font weight. Not sure why a) happens as the code seems to be fine?
[image]

* in section 4.1.24.4.1, "default UART’S TX" -> "default UART's TX"
* in section 4.3.13 "make repeating alarms (by return non zero from the callback)", maybe "by returning a non zero value"?
* in section 4.1.24.4.11 "block until all the character has been sent", maybe "until the entire character"?
* in section 4.1.24.4.4 no period in second sentence, see my q on that

lurch, 2021-07-16:
> in section 4.4.3 on pico_binary_info there seems to be a) inconsistent spacing for the macros

Unfortunately the inconsistent spacing is just because of how asciidoctor-pdf word-wraps lines containing such long "words".

> b) inconsistent font weight.

Yup, that (along with the odd `\__`) is another one of my XLST bugs, probably caused by the double-underscores not being correctly escaped or something. https://raspberrypi.github.io/pico-sdk-doxygen/group__pico__binary__info.html shows how it _ought_ to look (where you can see that the `__bi_ptr_lineno_var_name` and `__bi_lineno_var_name.core` have also lost their leading underscores in the PDF version). Added to my backlog.

> in section 4.3.13 "make repeating alarms (by return non zero from the callback)", maybe "by returning a nonzero value"?

That's section 4.2.13. :wink:  Yeah I'm not sure what that should say, that's probably one for @kilograham to answer. "abstract that further (at the cost of a user structure to store the repeat delay in (which the alarm framework does not have space for)." also has mis-balanced parentheses.

> `adc_rrobin()` doesn't exist anywhere I think..?

Perhaps that was supposed to be `adc_set_round_robin()` ? (it's possible that the function got renamed during development, and the Doxygen slipped behind? :shrug: )

> in section 4.1.10.4.10, "Clear the output enable (i.e. set to input) Clear any output value." not sure, same for .11

Looking at https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_gpio/include/hardware/gpio.h#L409 and https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__gpio.html#ga8aa4741d93dc40bc728dbd3e41813ad1 we probably just want to add a full-stop after "Clear the output enable (i.e. set to input)".

> in section 4.1.11.3.6, the clicky links for i2c0 and i2c1 don't go anywhere

Yup, looks like another XLST bug - the `i2c0` should probably link to the `i2c0_inst` in section 4.1.11.1. ?
However even in the [HTML version](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__i2c.html#ga56c7844696c095a3ad088100df011fd2) it only lists `i2c0_inst` and not `i2c1_inst` so fixing this properly might need some deeper fiddling to the doxygen itself. More stuff for my backlog. (It _does_ work properly for e.g. the `uart0` and `uart1` in 4.1.24.4.7 so at least I know that a fix is theoretically possible! :slightly_smiling_face: )

The rest of your points seem fine.

> OK, from now on the uncontroversial ones will be lumped in the PR.

Go for it! :tada:

matiasilva, 2021-07-19:
OK, fantastic. I've put all the ones with checks in https://github.com/raspberrypi/pico-sdk/pull/517. The one left I'm not sure about is the one on alarms, which I'll leave for Graham to comment on.

matiasilva, 2021-07-19:
There's "other specialized applications that doesn’t want" in 2.3.3 which should be "that don't want", but I assume this is in a separate adoc file.

lurch, 2021-07-19:
> There's "other specialized applications that doesn’t want" in 2.3.3 which should be "that don't want", but I assume this is in a separate adoc file.

Thanks, just fixed that :+1:

lurch, 2022-04-06:
> probably caused by the double-underscores not being correctly escaped or something.

This part has now been fixed in our internal tooling, and will be visible in the next public databooks-release. The screenshot in the first image now (correctly) looks like:
[image]

However I won't close this issue yet as there's still the problem with i2c0 and i2c1 to investigate.

lurch, 2022-12-02:
Closing as invalid, as the SDK API docs are now only at https://raspberrypi.github.io/pico-sdk-doxygen/ and no longer included in the C-SDK PDF.

closed 2022-12-02 by lurch