raspberrypi/pico-feedback #179: need help | wont build, spent days

Geekume, 2021-07-16:
So far Pico sadly has brought me more pain than happiness. I have been getting an error where the toolchain just straight up doesn't work. and the sad part is it once did. and now it doesnt. 

[build] C:\Users\tjsun\Downloads\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:74:68: error: 'uintptr_t' undeclared (first use in this function); did you mean 'uint64_t'?
[build]    74 | #define hw_xor_alias_untyped(addr) ((void *)(REG_ALIAS_XOR_BITS | (uintptr_t)(addr)))
[build]       |                                                                    ^~~~~~~~~
[build] C:\Users\tjsun\Downloads\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:108:19: note: in expansion of macro 'hw_xor_alias_untyped'
[build]   108 |     *(io_rw_32 *) hw_xor_alias_untyped((volatile void *) addr) = mask;
[build]       |                   ^~~~~~~~~~~~~~~~~~~~
[build] In file included from C:\Users\tjsun\Downloads\pico-sdk\src\common\pico_stdlib\include/pico/stdlib.h:14,
[build]                  from C:\Users\tjsun\Desktop\folder\main.c:2:
[build] C:\Users\tjsun\Downloads\pico-sdk\src\rp2_common\hardware_uart\include/hardware/uart.h: In function 'uart_get_instance':
[build] C:\Users\tjsun\Downloads\pico-sdk\src\rp2_common\hardware_uart\include/hardware/uart.h:105:5: warning: implicit declaration of function 'static_assert'; did you mean 'hard_assert'? [-Wimplicit-function-declaration]
[build]   105 |     static_assert(NUM_UARTS == 2, "");
[build]       |     ^~~~~~~~~~~~~
[build]       |     hard_assert
[build] NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102020~1\bin\AR19DD~1.EXE' : return code '0x1'
[build] Stop.
[build] NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.exe"' : return code '0x2'
[build] Stop.
[build] NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.exe"' : return code '0x2'
[build] Stop.
[build] Build finished with exit code 2

this is just some of the errors that go on. I have been trying to get help from discord servers and more and yet i still couldn't get it figured out. Probably spent in total over a days worth of hours trying to figure it out but i just cant. if anyone at all knows whats going on I would love and appreciate the enlightenment. 

- Geekume

kilograham, 2021-07-16:
Moved into pico-feedback, as this is a user environment issue

kilograham, 2021-07-16:
"this used to work" - implies something changed. It does indeed look like your build environment has become in some way broken/messed up.

1) I suggested on discord yesterday, but you did not reply/acknowledge that you blow away your build/ directory and recreate
2) when you do recreate and re-run `cmake` please record the output (and the command line you type) so you can post it here
3) Are you sure you are running the build from a VS Developer Command prompt as per the windows instructions?
4)  If the build fails, also include details on what you did to start the build (e.g. what directory you were in, what commands tou typed)

Geekume, 2021-07-16:
> "this used to work" - implies something changed. It does indeed look like your build environment has become in some way broken/messed up.
> 
> 1. I suggested on discord yesterday, but you did not reply/acknowledge, the first thing to try is to blow away your `build/` directory and recreate
> 2. when you do recreate by running `cmake` please record the output (and the command line you type) so you can post it here
> 3. Are you sure you are running the build from a VS Developer Command prompt as per the windows instructions?
> 4. If the build fails, also include details on what you did to start the build (e.g. what directory you were in, what commands tou typed)

I didn't do any of this in a terminal this in a terminal. Also i did not see anything on discord at all. also I have "blown away" the build folder like 500 times. thats not the issue. to answer 3) yes I did use the vs code developer prompt in admin. and to answer 4 i was in C:\Users\tjsun\Desktop\folder and opened the folder in VS Code. and it configures ok it says then i build it it generates the build folder then it gives me that error

Geekume, 2021-07-16:
> Have you tried the third party Windows installer? https://github.com/ndabas/pico-setup-windows

I had tried it and tried doing the build process even changing the path to the cloned sdk within the windows setup folder. I uninstalled and reinstalled the arm compiler for the 5th time only this time with the setup in the installers folder in the windows setup folder and still no luck. dude i dont even know whats going on anymore

kilograham, 2021-07-16:
> I didn't do any of this in a terminal this in a terminal

Ah well, I don't think that was obvious before. It sounds a lot like your Visual Studio set up is broken, so perhaps try building from the command line first. (i.e. instructions in section 9.2.3 of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf)

> The weird part too was even when i tried it on Arduino and get it set up with the official plugin for rp pico boards I was having similar issues. Weird question is there configs on the board itself that stays on there. It's just weird that the issue carries over when the plugin has their own arm tools

No there is no on board config... 

Did your Pico setup work *until* you installed the Arduino tools?

Geekume, 2021-07-16:
> Ah well, I don't think that was obvious before. It sounds a lot like your Visual Studio set up is broken, so perhaps try building from the command line first. (i.e. instructions in section 9.2.3 of [...]

Man I must of seen about a dozen people send me this link to the PDF but  unfortunately it's not a common issue

> No there is no on board config... 
There is a plugin for rp pico by Arduino themselves
> Did your Pico setup work *until* you installed the Arduino tools? 
There is not corralation between that and the vs code messing up it was broken before installed the plugin and when I did a project before when it worked I had Arduino with a 3rd party plugin installed worked fine back then.

Geekume, 2021-07-16:
[image]

Geekume, 2021-07-17:
If you don't have any answer for it mind putting it back on pico-sdk issues?

aallan, 2021-07-17:
Merged and closed #180 as duplicate.

kilograham, 2021-07-17:
You still haven't posted the output from the fresh `cmake` step as asked.

The compilation errors look like you are missing things defined in certain standard C library headers... this is certainly odd unless you have a broken arm-none-eabi-gcc install, or your project somehow has headers of the same name hiding the real ones.

Geekume, 2021-07-17:
What do you mean that's all that I got man the cmake config is fine it all passes it's just when nmake is introduced when its messed up

Geekume, 2021-07-17:
he

> You still haven't posted the output from the fresh `cmake` step as asked.
>
> The compilation errors look like you are missing things defined in certain standard C library headers... this is certainly [...]

here is the output for cmake c:\Pico\pico-examples\build>cmake -G "NMake Makefiles" ..
Using PICO_SDK_PATH from environment ('..\..\pico-sdk')
PICO_SDK_PATH is C:/Pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe
Defaulting PICO target board to pico since not specified.
Using board configuration from C:/Pico/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: C:/Users/tjsun/AppData/Local/Programs/Python/Python39/python.exe (found version "3.9.6") found compone
nts: Interpreter
TinyUSB available at C:/Pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Pico/pico-examples/build

kilograham, 2021-07-17:
what does `nmake VERBOSE=1` print

Geekume, 2021-07-17:
it prints 

c:\Pico\pico-examples\build>nmake VERBROSE=1

Microsoft (R) Program Maintenance Utility Version 14.29.30038.1
Copyright (C) Microsoft Corporation.  All rights reserved.

[  0%] Built target bs2_default
[  0%] Performing build step for 'ELF2UF2Build'

Microsoft (R) Program Maintenance Utility Version 14.29.30038.1
Copyright (C) Microsoft Corporation.  All rights reserved.

[100%] Built target elf2uf2
[  0%] No install step for 'ELF2UF2Build'
[  0%] Completed 'ELF2UF2Build'
[  0%] Built target ELF2UF2Build
[  0%] Built target bs2_default_padded_checksummed_asm
[  0%] Building C object blink/CMakeFiles/blink.dir/blink.c.obj
In file included from C:\Pico\pico-sdk\src\rp2040\hardware_structs\include/hardware/structs/timer.h:10,
                 from C:\Pico\pico-sdk\src\rp2_common\hardware_timer\include/hardware/timer.h:11,
                 from C:\Pico\pico-sdk\src\common\pico_time\include/pico/time.h:11,
                 from C:\Pico\pico-sdk\src\common\pico_stdlib\include/pico/stdlib.h:12,
                 from C:\Pico\pico-examples\blink\blink.c:7:
C:\Pico\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h: In function 'hw_set_bits':
C:\Pico\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:72:68: error: 'uintptr_t' undeclared (fi
rst use in this function); did you mean 'uint64_t'?
   72 | #define hw_set_alias_untyped(addr) ((void *)(REG_ALIAS_SET_BITS | (uintptr_t)(addr)))
      |                                                                    ^~~~~~~~~
C:\Pico\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:88:19: note: in expansion of macro 'hw_s
et_alias_untyped'
   88 |     *(io_rw_32 *) hw_set_alias_untyped((volatile void *) addr) = mask;
      |                   ^~~~~~~~~~~~~~~~~~~~
C:\Pico\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:72:68: note: each undeclared identifier
is reported only once for each function it appears in
   72 | #define hw_set_alias_untyped(addr) ((void *)(REG_ALIAS_SET_BITS | (uintptr_t)(addr)))
      |                                                                    ^~~~~~~~~
C:\Pico\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:88:19: note: in expansion of macro 'hw_s
et_alias_untyped'
   88 |     *(io_rw_32 *) hw_set_alias_untyped((volatile void *) addr) = mask;
      |                   ^~~~~~~~~~~~~~~~~~~~
C:\Pico\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h: In function 'hw_clear_bits':
C:\Pico\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:73:70: error: 'uintptr_t' undeclared (fi
rst use in this function); did you mean 'uint64_t'?
   73 | #define hw_clear_alias_untyped(addr) ((void *)(REG_ALIAS_CLR_BITS | (uintptr_t)(addr)))
      |                                                                      ^~~~~~~~~
C:\Pico\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:98:19: note: in expansion of macro 'hw_c
lear_alias_untyped'
   98 |     *(io_rw_32 *) hw_clear_alias_untyped((volatile void *) addr) = mask;
      |                   ^~~~~~~~~~~~~~~~~~~~~~
C:\Pico\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h: In function 'hw_xor_bits':
C:\Pico\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:74:68: error: 'uintptr_t' undeclared (fi
rst use in this function); did you mean 'uint64_t'?
   74 | #define hw_xor_alias_untyped(addr) ((void *)(REG_ALIAS_XOR_BITS | (uintptr_t)(addr)))
      |                                                                    ^~~~~~~~~
C:\Pico\pico-sdk\src\rp2_common\hardware_base\include/hardware/address_mapped.h:108:19: note: in expansion of macro 'hw_
xor_alias_untyped'
  108 |     *(io_rw_32 *) hw_xor_alias_untyped((volatile void *) addr) = mask;
      |                   ^~~~~~~~~~~~~~~~~~~~
In file included from C:\Pico\pico-sdk\src\common\pico_stdlib\include/pico/stdlib.h:14,
                 from C:\Pico\pico-examples\blink\blink.c:7:
C:\Pico\pico-sdk\src\rp2_common\hardware_uart\include/hardware/uart.h: In function 'uart_get_instance':
C:\Pico\pico-sdk\src\rp2_common\hardware_uart\include/hardware/uart.h:105:5: warning: implicit declaration of function '
static_assert'; did you mean 'hard_assert'? [-Wimplicit-function-declaration]
  105 |     static_assert(NUM_UARTS == 2, "");
      |     ^~~~~~~~~~~~~
      |     hard_assert
NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102020~1\bin\AR19DD~1.EXE' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bi
n\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bi
n\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.

c:\Pico\pico-examples\build>

kilograham, 2021-07-17:
VERBOSE not VERBROSE

Geekume, 2021-07-17:
c:\Pico\pico-examples\build>nmake VERBOSE=1

Microsoft (R) Program Maintenance Utility Version 14.29.30038.1
Copyright (C) Microsoft Corporation.  All rights reserved.

        "C:\Program Files\CMake\bin\cmake.exe" -SC:\Pico\pico-examples -BC:\Pico\pico-examples\build --check-build-syste
m CMakeFiles\Makefile.cmake 0
        "C:\Program Files\CMake\bin\cmake.exe" -E cmake_progress_start C:\Pico\pico-examples\build\CMakeFiles C:\Pico\pi
co-examples\build\\CMakeFiles\progress.marks
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.
exe"  -f CMakeFiles\Makefile2 /nologo -                   all
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.
exe"  -f pico-sdk\src\rp2_common\boot_stage2\CMakeFiles\bs2_default.dir\build.make /nologo -L                  pico-sdk\
src\rp2_common\boot_stage2\CMakeFiles\bs2_default.dir\depend
        "C:\Program Files\CMake\bin\cmake.exe" -E cmake_depends "NMake Makefiles" C:\Pico\pico-examples C:\Pico\pico-sdk
\src\rp2_common\boot_stage2 C:\Pico\pico-examples\build C:\Pico\pico-examples\build\pico-sdk\src\rp2_common\boot_stage2
C:\Pico\pico-examples\build\pico-sdk\src\rp2_common\boot_stage2\CMakeFiles\bs2_default.dir\DependInfo.cmake --color=
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.
exe"  -f pico-sdk\src\rp2_common\boot_stage2\CMakeFiles\bs2_default.dir\build.make /nologo -L                  pico-sdk\
src\rp2_common\boot_stage2\CMakeFiles\bs2_default.dir\build
[  0%] Built target bs2_default
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.
exe"  -f blink\CMakeFiles\ELF2UF2Build.dir\build.make /nologo -L                  blink\CMakeFiles\ELF2UF2Build.dir\depe
nd
        "C:\Program Files\CMake\bin\cmake.exe" -E cmake_depends "NMake Makefiles" C:\Pico\pico-examples C:\Pico\pico-exa
mples\blink C:\Pico\pico-examples\build C:\Pico\pico-examples\build\blink C:\Pico\pico-examples\build\blink\CMakeFiles\E
LF2UF2Build.dir\DependInfo.cmake --color=
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.
exe"  -f blink\CMakeFiles\ELF2UF2Build.dir\build.make /nologo -L                  blink\CMakeFiles\ELF2UF2Build.dir\buil
d
[  0%] Performing build step for 'ELF2UF2Build'
        cd C:\Pico\pico-examples\build\elf2uf2
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.
exe"

Microsoft (R) Program Maintenance Utility Version 14.29.30038.1
Copyright (C) Microsoft Corporation.  All rights reserved.

        "C:\Program Files\CMake\bin\cmake.exe" -SC:\Pico\pico-sdk\tools\elf2uf2 -BC:\Pico\pico-examples\build\elf2uf2 --
check-build-system CMakeFiles\Makefile.cmake 0
        "C:\Program Files\CMake\bin\cmake.exe" -E cmake_progress_start C:\Pico\pico-examples\build\elf2uf2\CMakeFiles C:
\Pico\pico-examples\build\elf2uf2\\CMakeFiles\progress.marks
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.
exe"  -f CMakeFiles\Makefile2 /nologo -                   all
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.
exe"  -f CMakeFiles\elf2uf2.dir\build.make /nologo -L                  CMakeFiles\elf2uf2.dir\depend
        "C:\Program Files\CMake\bin\cmake.exe" -E cmake_depends "NMake Makefiles" C:\Pico\pico-sdk\tools\elf2uf2 C:\Pico
\pico-sdk\tools\elf2uf2 C:\Pico\pico-examples\build\elf2uf2 C:\Pico\pico-examples\build\elf2uf2 C:\Pico\pico-examples\bu
ild\elf2uf2\CMakeFiles\elf2uf2.dir\DependInfo.cmake --color=
        "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.
exe"  -f CMakeFiles\elf2uf2.dir\build.make /nologo -L                  CMakeFiles\elf2uf2.dir\build
[100%] Built target elf2uf2
        "C:\Program Files\CMake\bin\cmake.exe" -E cmake_progress_start C:\Pico\pico-examples\build\elf2uf2\CMakeFiles 0
        cd C:\Pico\pico-examples\build
[  0%] No install step for 'ELF2UF2Build'
        cd C:\Pico\pico-examples\build\elf2uf2
        echo >nul && "C:\Program Files\CMake\bin\cmake.exe" -E echo_append
        echo >nul && "C:\Program Files\CMake\bin\cmake.exe" -E touch C:/Pico/pico-examples/build/blink/elf2uf2/src/ELF2U
F2Build-stamp/ELF2UF2Build-install
        cd C:\Pico\pico-examples\build
[  0%] Completed 'ELF2UF2Build'
        cd C:\Pico\pico-examples\build\blink
        echo >nul && "C:\Program Files\CMake\bin\cmake.exe" -E make_directory C:/Pico/pico-examples/build/blink/CMakeFil
es
        echo >nul && "C:\Program Files\CMake\bin\cmake.exe" -E touch C:/Pico/pico-examples/build/blink/CMakeFiles/ELF2UF
2Build-complete
        echo >nul && "C:\Program Files\CMake\bin\cmake.exe" -E touch C:/Pico/pico-examples/build/blink/elf2uf2/src/ELF2U
F2Build-stamp/ELF2UF2Build-done
        cd C:\Pico\pico-examples\build
        cd C:\Pico\pico-examples\build\blink
[... 84 lines cut ...]
rst use in this function); did you mean 'uint64_t'?
   74 | #define hw_xor_alias_untyped(addr) ((void *)(REG_ALIAS_XOR_BITS | (uintptr_t)(addr)))
      |                                                                    ^~~~~~~~~
[... cut ...]
      |     hard_assert
NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102020~1\bin\AR19DD~1.EXE' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bi
n\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bi
n\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.

c:\Pico\pico-examples\build>

kilograham, 2021-07-17:
That looks OK, perhaps something in your environment then screwing up GCC (it is not picking up its own header files correctly)

can you list you environment variables with `SET` in your developer command prompt.

Geekume, 2021-07-17:
ok here you go 

c:\Pico\pico-examples\build>SET
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\tjsun\AppData\Roaming
CommandPromptType=Native
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPILER_PATH=C:\psyq\bin
COMPUTERNAME=DESKTOP-CHON5GA
ComSpec=C:\WINDOWS\system32\cmd.exe
C_INCLUDE_PATH=C:\psyq\include
C_PLUS_INCLUDE_PATH=C:\psyq\include
DevEnvDir=C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\Common7\IDE\
DriverData=C:\Windows\System32\Drivers\DriverData
ExtensionSdkDir=C:\Program Files (x86)\Microsoft SDKs\Windows Kits\10\ExtensionSDKs
Framework40Version=v4.0
FrameworkDir=C:\Windows\Microsoft.NET\Framework\
FrameworkDir32=C:\Windows\Microsoft.NET\Framework\
FrameworkVersion=v4.0.30319
FrameworkVersion32=v4.0.30319
G032TMP=C:\TEMP
GO32=DPMISTACK 1000000
HAXEPATH=C:\HaxeToolkit\haxe
HOMEDRIVE=C:
HOMEPATH=\Users\tjsun
INCLUDE=C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\include;C:\Program File
s (x86)\Windows Kits\10\include\10.0.19041.0\ucrt;C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared;C:\
Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um;C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\
winrt;C:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt
LIB=C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\lib\x86;C:\Program Files (x
86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x86;C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\um\x86
LIBPATH=C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\lib\x86;C:\Program File
s (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\lib\x86\store\references;C:\Program Files (x86
)\Windows Kits\10\UnionMetadata\10.0.19041.0;C:\Program Files (x86)\Windows Kits\10\References\10.0.19041.0;C:\Windows\M
icrosoft.NET\Framework\v4.0.30319
LIBRARY_PATH=C:\psyq\lib
LOCALAPPDATA=C:\Users\tjsun\AppData\Local
LOGONSERVER=\\DESKTOP-CHON5GA
MEmu_Path=C:\Program Files (x86)\Microvirt
NEKO_INSTPATH=C:\HaxeToolkit\neko
NUMBER_OF_PROCESSORS=6
OneDrive=C:\Users\tjsun\OneDrive
OneDriveConsumer=C:\Users\tjsun\OneDrive
OS=Windows_NT
Path=C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86;C:\Program
 Files (x86)\Microsoft Visual Studio\2019\BuildTools\Common7\IDE\VC\VCPackages;C:\Program Files (x86)\Microsoft Visual S
tudio\2019\BuildTools\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files (x86)\Microsoft Visual Studio\2
019\BuildTools\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files (x86)\Microsoft Visu
al Studio\2019\BuildTools\MSBuild\Current\bin\Roslyn;C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\Comm
on7\Tools\devinit;C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x86;C:\Program Files (x86)\Windows Kits\10\bin
\x86;C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\\MSBuild\Current\Bin;C:\Windows\Microsoft.NET\Framew
ork\v4.0.30319;C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\Common7\IDE\;C:\Program Files (x86)\Micros
oft Visual Studio\2019\BuildTools\Common7\Tools\;C:\Users\tjsun\AppData\Local\Programs\Python\Python39\Scripts;C:\Users\
tjsun\AppData\Local\Programs\Python\Python39;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\iCLS\;C:
\Program Files\Intel\Intel(R) Management Engine Components\iCLS\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem
;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\Intel\Intel(R) Manageme
nt Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\NVIDIA
Corporation\PhysX\Common;C:\Program Files (x86)\Brackets\command;C:\Program Files (x86)\GnuWin32\bin;C:\MinGW\bin;C:\WIN
DOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSS
H\;C:\HaxeToolkit\haxe;C:\HaxeToolkit\neko;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin;C:\Use
rs\tjsun\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Visual Studio Code;C:\Program Files\PuTTY\;C:\Program Fil
es\CMake\bin;C;C;C;C:\Program Files\Git\cmd;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin;C:\ps
yq\bin;;C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\Common7\IDE\CommonExtensions\Microsoft\CMake\CMak
e\bin;C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
PICO_SDK_PATH=..\..\pico-sdk
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 158 Stepping 10, GenuineIntel
[... 38 lines cut ...]
WindowsLibPath=C:\Program Files (x86)\Windows Kits\10\UnionMetadata\10.0.19041.0;C:\Program Files (x86)\Windows Kits\10\
References\10.0.19041.0
WindowsSdkBinPath=C:\Program Files (x86)\Windows Kits\10\bin\
WindowsSdkDir=C:\Program Files (x86)\Windows Kits\10\
WindowsSDKLibVersion=10.0.19041.0\
WindowsSdkVerBinPath=C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\
WindowsSDKVersion=10.0.19041.0\
WWISEROOT=C:\Program Files (x86)\Audiokinetic\Wwise 2021.1.1.7601
WWISESDK=C:\Program Files (x86)\Audiokinetic\Wwise 2021.1.1.7601\SDK
[... cut ...]
Program Files (x86)\Brackets\command;C:\Program Files (x86)\GnuWin32\bin;C:\MinGW\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\
WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\HaxeToolkit\haxe;C:\Ha
xeToolkit\neko;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin;C:\Users\tjsun\AppData\Roaming\Mic
rosoft\Windows\Start Menu\Programs\Visual Studio Code;C:\Program Files\PuTTY\;C:\Program Files\CMake\bin;C;C;C;C:\Progra
m Files\Git\cmd;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin;C:\psyq\bin;

c:\Pico\pico-examples\build>

kilograham, 2021-07-18:
```
C_INCLUDE_PATH=C:\psyq\include
C_PLUS_INCLUDE_PATH=C:\psyq\include
```

could be bad.

Geekume, 2021-07-18:
> What do you mean that's all that I got man the cmake config is fine it all passes it's just when nmake is introduced when its messed up and the fresh start gives the exact same thing so there would be no point it would be rescreenshotting the same thing

I mean I have those but what should I it be any substitution?

kilograham, 2021-07-18:
Unset them completely and try again

Geekume, 2021-07-18:
it worked thank you once I got rid of those it built normally. Now the frustration is over I appreciate your patience 
[image]

closed 2021-07-18 by Geekume

JamesH65, 2021-07-18:
Hi, just so we can help people in the future, have you any idea how those settings got in there in the first place?

Wren6991, 2021-07-18:
Looks like a [3rd-party PlayStation SDK](http://www.psxdev.net/help/psyq_install.html) which added itself to global include paths when installed, and screwed up other C SDKs in the process?

Geekume, 2021-07-19:
> Hi, just so we can help people in the future, have you any idea how those settings got in there in the first place?

It's if you have a path unrelated to the pico sdk path a the c compiler path the issue really persist on programmers that dabble in a lot of things

kilograham, 2021-07-19:
Well this is an interaction with GCC not the SDK. But it is bad practice by tools to install such paths globally (because invariable they will affect other things)