raspberrypi/pico-feedback #184: Incorrect display of library subgroup in SDK PDF

kilograham, 2021-08-01:
subgroups in the doxygen are being shown thusly:

[image]

Apparently this was deliberate, but we should therefore change the section headings (not other links in the documentation necessarily) to be _pico_sync (mutex)_ or similar

Note, I noticed this in response to this https://www.raspberrypi.org/forums/viewtopic.php?f=145&t=316893#p1896179 request for the header names, and I thought the answer was well if the library is called `pico_sync`, then the header is `"pico/sync.h"` which is how it is designed to but these toplevel headers massively confuse this issue (and don't make much sense anyway)

kilograham, 2021-08-01:
perhaps a (better?) alternative would just make those sub section headings one level lower themselves instead (without moving all their contents any lower)

kilograham, 2021-08-01:
or just insert a horizontal line or something and a label

lurch, 2021-08-01:
> perhaps a (better?) alternative would just make those sub section headings one level lower themselves instead (without moving all their contents any lower)

That would be ideal (and would match up with the [HTML SDK docs](https://raspberrypi.github.io/pico-sdk-doxygen/modules.html)), but AFAIK the ToC-levels in AsciiDoctor are (only) generated directly from the header-levels in the contents. https://docs.asciidoctor.org/asciidoc/latest/toc/

lurch, 2022-06-16:
Just adding a reminder for future-lurch that the ordering of these sections is determined by https://github.com/raspberrypi/pico-sdk/blob/master/docs/index.h (and then all the Doxygen automagic stuff that drops out of that).

lurch, 2022-12-02:
Closing as invalid, as the SDK API docs are now only at https://raspberrypi.github.io/pico-sdk-doxygen/ and no longer included in the C-SDK PDF.

closed 2022-12-02 by lurch