raspberrypi/pico-feedback #187: Suggestion for SDK documentation...

PhillHS, 2021-08-11:
Hi there,

It might be a good idea to add a note to which header file the functions are defined in at the beginning of each manual section in the docs.
This means you know which header files to include to get access to the functions.

Cheers.

Phill.

lurch, 2021-08-11:
I assume you're talking about the documentation in https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf ?
That is modelled after e.g. https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__dma.html which also doesn't mention which header file you need to include?
Any ideas @kilograham ?  (I'll assign this to myself, since any change here would probably need tweaks to the (internal) XSLT file we use to auto-generate parts of raspberry-pi-pico-c-sdk.pdf )

Hmmm, have you read section 2.4 of https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf ?

kilograham, 2021-08-11:
> Hmmm, have you read section 2.4 of https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf ?

This should be entirely the answer, however this is the sub-header issue with things like mutex that i was mentioning a while back.

lurch, 2021-08-11:
> this is the sub-header issue with things like mutex that i was mentioning a while back.

Right. Should I close this as a duplicate of #184 ?

lurch, 2022-12-02:
Closing as invalid, as the SDK API docs are now only at https://raspberrypi.github.io/pico-sdk-doxygen/ and no longer included in the C-SDK PDF.

closed 2022-12-02 by lurch